microscope-cockpit/cockpit #672: Camera wavelengths do not work with multiple filter wheels

dstoychev, 2020-08-24:
When multiple filter wheels are used, it seems that only the cameras of the last one have their wavelengths properly indicated in the user interface. Here are configs for a small working example

microscope:
```python
#!/usr/bin/python
"""Configuratin file for microscope.deviceserver.

Import device classes, then define entries in DEVICES as:
   devices(CLASS, HOST, PORT, other_args)
"""
# Function to create record for each device.
from microscope.devices import device
# Import device modules/classes here.
import microscope.testsuite.devices as testdevices

DEVICES = [
    device(testdevices.TestCamera, '127.0.0.1', 8000),
    device(testdevices.TestCamera, '127.0.0.1', 8001),
    device(testdevices.TestCamera, '127.0.0.1', 8002),
    device(testdevices.TestFilterWheel, '127.0.0.1', 8003, {'positions': 3}),
    device(testdevices.TestFilterWheel, '127.0.0.1', 8004, {'positions': 3})
]
```

cockpit
```
[server]
ipAddress: 127.0.0.1

[camera-test-red]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:TestCamera@127.0.0.1:8000

[camera-test-green]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:TestCamera@127.0.0.1:8001

[camera-test-blue]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:TestCamera@127.0.0.1:8002

[filter-wheel-1]
type: cockpit.devices.microscopeDevice.MicroscopeFilter
uri: PYRO:TestFilterWheel@127.0.0.1:8003
cameras: camera-test-red camera-test-green
filters:
    0, Alexa Fluor 488, 525
    1, RFP, 590
    2, Cy5, 670

[filter-wheel-2]
type: cockpit.devices.microscopeDevice.MicroscopeFilter
uri: PYRO:TestFilterWheel@127.0.0.1:8004
cameras: camera-test-blue
filters:
    0, DAPI, 460
    1, mCherry, 610
    2, Cy7, 780
```

dstoychev, 2020-08-25:
The values of the ``cameras`` key must be separated by coma or semicolon. Closing.

closed 2020-08-25 by dstoychev

carandraug, 2020-08-25:
Do you think we can improve the situation by having some sort of checks for this isse? Or maybe also include space as a separator for values of `cameras`?

I'm guessing the reason to not use space as separator is that space is a valid character in device name (commas and semicolon are also valid but maybe they're less useful)?

dstoychev, 2020-08-25:
I think that I used a space here because I was looking at the objective config recently. If I am not mistaken, space separators are valid there or at least the parsing is much more permissive. In general, as long as the config syntax is documented problems like this can be easily avoided. Or at least more examples. I don't think adding space would improve the situation. It makes much more sense to me that items are separated by proper punctuation such as comma or semicolon.

carandraug, 2020-08-28:
> In general, as long as the config syntax is documented problems like this can be easily avoided. Or at least more examples. I don't think adding space would improve the situation. It makes much more sense to me that items are separated by proper punctuation such as comma or semicolon.

I agree with all of that. Could you write that documentation please, then? The last one to be documented was `ObjectiveDevice`, you could use it as an example.

I think it might make sense to have some step where devices that require others can validate themselves. If there is no camera with the given name, there should be some error or at least an error. If that existed, you'd have got an error saying that there was no camera named `"camera-test-red camera-test-green"` and things would have been clearer. Maybe during `Device.finalizeInitialization`?