microscope-cockpit/cockpit #673: branches for site specific code/configuration

carandraug, 2020-08-28:
Currently we have the following branches in git, which I don't think will ever be merged in master.

* "deepsim", "cryosim", adn "diamondfixes": these three branches are about site specific configuration. Their configurations are no longer up to date and even if they were, this is not the place to keep them (we have the cockpit-configs repository). In addition, the "deepsim" branch also has `cockpit/devices/alpaoAO-labview.py` but I understand that's not something we want to continue, and the "cryosim" branch has "cockpit.py" and "cockpiy.pyw" which were the scripts to start cockpit which now are created by setuptools.

* "OMXv2" branch has mainly changes to `cockpit/devices/nanomover.py` and `cockpit/devices/ni6036e.py`. @iandobbie are the changes to `nanomover.py` worth keeping if the plan is to have it on microscope? The `ni6036e` module has already been removed.

* "nanomover-display-fix" is a branch of the "OMXv2" branch which has more changes to `nanomover.py`

I propose we just remove them (@iandobbie for has both the "OMXv2" and "nanomover-display-fix" branches for future references if they are ever needed).

iandobbie, 2020-09-02:
You are correct the site specific configs should be in the configs repository. So I think we should just delete these branches, but maybe let me make sure I have the latest versions before you do this.

carandraug, 2020-09-16:
I just checked and you only have the "nanomover-display-fix" and "OMXv2" branches, not the other three. Can you check the other ones?

Note that you do have a "deepsim" branch (and a "DeepSIM" one as well), bit neither is related to the "deepsim" branch of MicronOxford. You also have a "cryosim-working" (which is also unrelated to the "cryosim" branch)

iandobbie, 2020-09-17:
I have graded the relevant branches and deleted them from the micronoxford github repository so closing issue.

closed 2020-09-17 by iandobbie