microscope-cockpit/cockpit #674: cockpit never exits cleanly

carandraug, 2020-08-29:
When cockpit is closed, there is always an error about destroying the GUI. This is when we are destroying everything so we never see this error on the logging window. The following can be seen on the log file:

```
2020-08-29 20:52:31,006 ERROR      __init__: 271  Destroying <cockpit.gui.loggingWindow.LoggingWindow object at 0x7fe6022aaca8>
2020-08-29 20:52:31,006 ERROR      __init__: 271  Destroying <wx._core.Frame object at 0x7fe5f01c1288>
2020-08-29 20:52:31,007 ERROR      __init__: 271  Destroying <cockpit.gui.mainWindow.MainWindow object at 0x7fe60012f1f8>
2020-08-29 20:52:31,007 ERROR      __init__: 271  Destroying <cockpit.gui.camera.window.CamerasWindow object at 0x7fe6000ea8b8>
2020-08-29 20:52:31,007 ERROR      __init__: 271  Destroying <cockpit.gui.mosaic.window.MosaicWindow object at 0x7fe6000eaf78>
2020-08-29 20:52:31,007 ERROR      __init__: 271  Destroying <cockpit.gui.macroStage.macroStageWindow.MacroStageWindow object at 0x7fe5f2f45708>
2020-08-29 20:52:31,007 ERROR      __init__: 271  Destroying <cockpit.gui.shellWindow.ShellWindow object at 0x7fe5f2f745e8>
2020-08-29 20:52:31,007 ERROR      __init__: 271  Destroying <cockpit.gui.touchscreen.TouchScreenWindow object at 0x7fe5f038ed38>
2020-08-29 20:52:31,007 ERROR      __init__: 271  Destroying <cockpit.util.intensity.IntensityProfilerFrame object at 0x7fe5f0243b88>
2020-08-29 20:52:31,007 ERROR      __init__: 296  Still have non-daemon threads <map object at 0x7fe5f00d0fd0>
2020-08-29 20:52:31,007 ERROR      __init__: 298  {'_target': None, '_name': 'MainThread', '_args': (), '_kwargs': {}, '_daemonic': False, '_ident': 140626290829120, '_tstate_lock': <locked _thread.lock object at 0x7fe61bc11198>, '_started': <threading.Event object at 0x7fe61bc0cb70>, '_is_stopped': False, '_initialized': True, '_stderr': <_io.TextIOWrapper name='<stderr>' mode='w' encoding='UTF-8'>}
```

To replicate, it's enough to start and close cockpit, the check the log file.

carandraug, 2020-08-29:
This all comes from `CockpitApp.onExit` (note this is not `OnExit` so different capitalisation from the method that wx expects to do the cleanup).

The thread that stays at the end is the main thread. I just fixed that by ignoring the main thread when checking for remaining threads.

I believe that the issue with the other windows is because we are running `onExit` before `OnExit` which is when wx handles their destruction. There might be no issue (other than we never let wx cleanup because we call `os._exit` at the end of `onExit`. Need to investigate further.

carandraug, 2020-08-31:
I think I have tracked down all these issues now. Closing as fixed. Details of all the issues follow.

The windows mentioned on the error had not been destroyed before because wx has not got a chance at it. That should have happened during App's `OnExit`. When I fixed the method capitalization, then there was no leftover windows. This brought some issues:

1. since there is no `LoggingWindow`, then any errors and messages during this stage fail because the `LoggingWindow` hijacks stdout and stderr. Fixed with 865c2db4128030d46cf4ce8d18295ce51e0acda4
2. since the windows no longer exits during `OnExit`, we can't save the window position, so I moved that to `MainWindow.OnClose`
3. windows subscribe to events but never unsubscribe after they are destroyed. This is so far not much of a problem because cockpit keeps the windows until the end. This was now a problem because the mosaic window subscribes to `USER_ABORT` which is published during `OnExit` after the window has been destroyed. Fixed this by using `EvtEmitter` in 7d797b67a50b76509b158814d4d8f6cf18c0b84c. We should probably be doing this for all event subscriptions in all windows.

I have also found that the check for running threads could be moved out of `OnExit` to after `MainLoop` returns, and I made the `os._exit` conditional (only called if we find bad threads). This means that for the first time in years, Python is getting a chance at cleaning up which brought other issues:

1. the Pyro proxies need to be released, otherwise we get:
```
Exception ignored in: <function Proxy.__del__ at 0x7fcc5c2a3598>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/Pyro4/core.py", line 266, in __del__
  File "/usr/lib/python3/dist-packages/Pyro4/core.py", line 400, in _pyroRelease
  File "/usr/lib/python3.7/logging/__init__.py", line 1370, in debug
  File "/usr/lib/python3.7/logging/__init__.py", line 1626, in isEnabledFor
TypeError: 'NoneType' object is not callable
```
I fixed this with 19af3334a56ac28b198aa3ae5b9a2e56a99f5227

2. Cockpit was always exiting with a `Segmentation fault`. This was an issue on `EvtEmitter` and I fixed it with 2daeb3927dc65c8401bd405018792341cc638b05

closed 2020-08-31 by carandraug

iandobbie, 2020-09-01:
Well done. I knew there was an exit problem  but always put of investigating.