microscope-cockpit/cockpit #675: Touchscreen view mosaic tile issue.

iandobbie, 2020-09-08:
I think that the touchscreen view overwrites the current megatile when you open it and start to collect data from it. 

How to trigger this bug:
1) start cockpit with touchscreen disabled
2) turn on a light and camera and collect a small mosaic
3) enable the touchscreen from the windows menu - see the pixel data s there at both low and high zooms.
4) notice that the pixel data is there
5) Continue the mosaic in the touchscreen

The existing data is still there a high zooms, ie in the normal tiles, but if you zoom out the previous mosaic data vanishes, ie the megatile has been overwritten.

iandobbie, 2021-03-09:
I'm afraid that the touchscreen code deals with the mosaic canvas in totally the worng way, which might be causing Ilan's issue, but also causes the megatile issues. 

The touchsceen has this in its init:
        self.canvas = cockpit.gui.mosaic.canvas.MosaicCanvas(
            self, limits, self.drawOverlay, self.onMouse
        )
 
And the the canvas code starts with this:

class MosaicCanvas(wx.glcanvas.GLCanvas):
    ## Tiles and context are shared amongst all instances, since all
    # offer views of the same data.
    # The first instance creates the context.
    ## List of MegaTiles. These will be created in self.initGL.
    megaTiles = []
    ## List of Tiles. These are created as we receive new images from
    # our parent.
    tiles = []
    ## Set of tiles that need to be rerendered in the next onPaint call.

The comments in the code are rendered false by the next line, the class generates local tiles and megaTile lists which are then populated within each instance. This is not how I was doing it and I am sure it is wrong as we are generating 2 copies of each of these large object lists.

iandobbie, 2021-03-09:
This can clearly be seen if you collect some data from the test camera, with random intensity ranges. Then do a rescale data which rescales the data to 8 bit with a global min/max intensity and uploads to the GPU. This will only change the "normal" mosaic view. Danny's touchscreen code is creating a second set of textures for bith tiles and megatiles. This should just be a second view into the same tile data.

carandraug, 2021-03-10:
I can reproduce this issue in Linux. After restarting the mosaic on the touchscreen. If I zoom out, in the macro stage or the touchscreen, the previously acquired tiles disappear.

> This is not how I was doing it and I am sure it is wrong as we are generating 2 copies of each of these large object lists.

When you say "how I was doing it", do you mean on the previous touchscreen? I just tested an old version of the touchscreen, from before we merged Danny's changes, and I can reproduce this issue there too.

iandobbie, 2021-03-10:
> When you say "how I was doing it", do you mean on the previous touchscreen? I just tested an old version of the touchscreen, from before we merged Danny's changes, and I can reproduce this issue there too.

Yes I was referring to the old touchscreen version, and obviously I didn't test this! Sorry for the incorrect bug reporting. Does seem to me that the two mosaic instances are creating separate tile instances, which seems wrong to me.