microscope-cockpit/cockpit #678: Rather strange deformable mirror error.

iandobbie, 2020-10-06:
I am trying to get the CryoSIM DM working and Nick cockpit module sets the mirror trigger to rising edge and microscope returns the following error

    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
microscope.DeviceError: PEX-292144 only support trigger on falling edge (error code 266)

My questions are, a) is this true? and b) How does Nick's code on DeepSIM work if this is the case?

carandraug, 2020-10-06:
From which device is this error coming from? Is this the `AlpaoDeformableMirror` using current main master? If so, and I get it right, is this the code to reproduce the problem (removing the device server from the equation)?

    from microscope.mirror.alpao import AlpaoDeformableMirrot
    from microscope import TriggerMode, TriggerType
    dm = AlpaoDeformableMirror(serial_number="???")
    dm.set_trigger(TriggerType.FALLING_EDGE, TriggerMode.ONCE)

iandobbie, 2020-10-14:
Sorry to answer these questions I need to be in the lab on the cryoSIM setup. I reply once I have some time there.

iandobbie, 2021-01-19:
I believe that I was implementing AOTools incorrectly, we probably need a guide to do this. Anyway it is now working so this was just user error.

closed 2021-01-19 by iandobbie