microscope-cockpit/cockpit #679: microscopeDeformableMirror assumes a square image for wavefront sensor.

iandobbie, 2020-10-06:
The microscopeDeformabelMirror composite device including a DM and an intereferometer for wavefront sensing assumes that the camera on the interferometer is square.

Can be worked around by setting a square ROI on the camera

carandraug, 2020-10-06:
None of of the deformable mirror stuff has been merged yet. Adding @NickHallONBI which I guess is the author of code one some of his branches.

NickHallONBI, 2020-10-06:
This is a Microscope-AOtools issue, not a cockpit issue. I will add it to the Microscope-AOtools project

closed 2020-10-06 by NickHallONBI

carandraug, 2020-10-06:
This is a link to the issue opened MicronOxford/microscope-aotools#5 (by the way, Nick, it's possible to transfer issues between repositories, see the "Transfer issue" button at the bottom of the right side bar).