microscope-cockpit/cockpit #682: Cameras do not show real on/off status at startup of cockpit

juliomateoslangerak, 2020-10-16:
If I leave the cameras on when I quit cockpit, the next time I start it they show as off. However I guess they are, in reality on cause I cannot change camera settings that require camera to be off.
I guess the cameras should be turned off at shutdown (maybe not), but in any case the GUI does not reflect the real status.

iandobbie, 2022-05-22:
This should be trivial to test with a call like

        #get real enable state.
        self.enabled=self._proxy.get_is_enabled()

Need to do this once the pyro connection has been established, maybe in init or maybe in initialise?

carandraug, 2022-05-23:
This is not as straightforward. There's a bunch of places where things don

1. `cockpit.gui.device.EnableButton` is constructed assuming that device is off. It should check.
2. but `EnableButton` does not just show whether the device is enabled or not, it has different colours for different states (states are one of `cockpit.handlers.deviceHandlers.STATES`) but there is no method to get the state of a handler, you can only subscribe to the `DEVICE_STATUS` event and wait for it to change.
  1. As an aside, we never actually the `STATES.disabled` and `STATES.enabled`. It seems we rely on the `bool` for `getIsEnabled` being converted to `1` or `0`.

3. We can get away from having `EnableButton` check the status at start (although I think it should) because handlers emit a `DEVICE_STATUS` event after everything is ready to start (published by `DeviceHandler.makeInitialPublications`). However:
  1. `CameraHandler` assumes it starts disabled so that publication emits that the camera is disabled when in fact is enabled.
  2. the `CameraHandler` method that returns the enable/disabled state uses a cache value which only changes when `setEnabled` is called
  3. when `MicroscopeCamera` constructs the camera handler it has no option to pass the current enable state, it is always assumed that it is constructed when the camera is disabled.
  4. the `MicroscopeCamera` itself is also constructed and set as disabled. It should check its status first.

iandobbie, 2022-05-23:
Oh well obviously a lot more involved than I thought. Seems like it involves a few steps.

1) check the status on connection, not just assume it.

2) ensure that the cache value is updated and correct

3) trigger an event to ensure the correct status is broadcast to any subscribed modules, eg touchscreen etc...

There will obviously need to be changes in a bunch of places to make this work and be coherent, but it should all be possible on the cockpit side as there is a well defined call to microscope to find th camera state.