microscope-cockpit/cockpit #683: Windows 10 menu bar windows menu problem.

iandobbie, 2020-10-22:
On windows 10 the windows menu isn't properly populated. It has reset window positons, Launch ValueLogViewer and a dsp debug window but none of the other windows are listed. This means it is hard to get back a lost or buried
[image]
 window, especially on a small screen.

carandraug, 2020-10-22:
I can't reproduce the issue with a Windows 10 VM and I've seen others using it on a real Windows 10. Are you using MicronOxford's master branch or some other master branch? Can you investigate what's happening on that system (see `cockpit.gui.mainWindow.WindowsMenu.OnMenuOpen`)?

iandobbie, 2020-10-22:
The branch is micronoxford/master as of this morning. I will investigate what happening on that call.

iandobbie, 2020-10-22:
Installed pycharm and played around a bit. As far as I can tell the OnMenuOpen is not getting called, hence the issue about it not populating the menu.

carandraug, 2020-10-22:
Check with the event watcher what is happening to it (it's pretty unreliable but may give some hints):

```
from wx.lib.inspection import InspectionTool
InspectionTool().Show() # then click on the big Events button in the tool bar
```

iandobbie, 2020-10-22:
From the wx documentation 

EVT_MENU_OPEN(func):
A menu is about to be opened. On Windows, this is only sent once for each navigation of the menubar (up until all menus have closed).

I don't understand why mousing over the menu and selecting it doesn't run this function but it isn't being called in pycharm.

iandobbie, 2020-10-22:
I wonder if the menu_open event only happens on the parent, I guess the menubar.

carandraug, 2020-10-24:
I got my hands on the machine that had this problem, and the problem seems to have been the wxPython version (it was running wxPython 3.0.6 with wxWidgets version 3.0.5).  I installed wxPython 4.1.0 and the problem was solved. Maybe we should just require wxPython 4.1.

iandobbie, 2020-10-24:
Well spotted. There is a wx issue on Mac that is still unsolved. We should make sure we don’t break the Mac version while fixing the pc one.

Ian

carandraug, 2020-10-24:
> There is a wx issue on Mac that is still unsolved.

Which one are you talking about? I thought maybe you meant #628 but I'm trying to replicate it on a Mac now and I'm not being successful.

iandobbie, 2020-10-27:
This issue was a wx version problem. Now solved

closed 2020-10-27 by iandobbie