microscope-cockpit/cockpit #684: 'NoneType' object is not callable exception from freetype at exit

carandraug, 2020-10-22:
Sometimes (maybe once every 10 times or so) when cockpit exits, the following error appears:

```
Exception ignored in: <function _Glyph.__del__ at 0x7f1b08898d08>
Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/gui/freetype.py", line 152, in __del__
  File "/usr/lib/python3/dist-packages/OpenGL/latebind.py", line 61, in __call__
  File "/usr/lib/python3/dist-packages/OpenGL/GL/exceptional.py", line 55, in glDeleteTextures
  File "/usr/lib/python3/dist-packages/OpenGL/arrays/arraydatatype.py", line 161, in arraySize
  File "/usr/lib/python3/dist-packages/OpenGL/arrays/arraydatatype.py", line 43, in __call__
  File "/usr/lib/python3/dist-packages/OpenGL/plugins.py", line 16, in load
  File "/usr/lib/python3/dist-packages/OpenGL/plugins.py", line 38, in importByName
TypeError: 'NoneType' object is not callable
```

This seems to happen during cleanup, when we're no longer writing to the log files, so this will not appear there (it only appears on the command line).

I have no clue why this happening (whatever little documentation PyOpenGL has is wrong). Things that I have checked:

* `glDeleteTextures` signature really is `glDeleteTextures(array_of_texture_ids)` and it is not `glDeleteTextures(n, textures)` (the documentation is wrong)
* texture_id being zero (I have checked and all textures being deleted have ID higher than zero)
* glDeleteTexture evalutes as True (no clue why this is needed but I have seen code snippets where that is checked)

I have also tried to upgrade PyOpenGl to 3.1.5 (latest on PyPI) and the error is the same (but the stack is different):

```
Exception ignored in: <function _Glyph.__del__ at 0x7f0a365f4b70>
Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/gui/freetype.py", line 151, in __del__
  File "/usr/local/lib/python3.7/site-packages/OpenGL/lazywrapper.py", line 23, in __bool__
  File "/usr/local/lib/python3.7/site-packages/OpenGL/wrapper.py", line 98, in __nonzero__
  File "/usr/local/lib/python3.7/site-packages/OpenGL/platform/baseplatform.py", line 383, in __nonzero__
  File "/usr/local/lib/python3.7/site-packages/OpenGL/platform/baseplatform.py", line 389, in load
TypeError: 'NoneType' object is not callable
```

ARCFaria, 2021-03-08:
I tried David's branch 684-texture-release, and got this error:
```
Exception ignored in: <function Face.__del__ at 0x7f7faa7c3160>
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/freetype/__init__.py", line 1140, in __del__
TypeError: 'NoneType' object is not callable
```

carandraug, 2021-03-09:
The branch that @ARCFaria refers to is my initial attempt at fixing this. It fixed the issue in Linux. I'm not sure where the root of the issue was but I was guessing that `Glyph.__del__` was being called after the OpenGL context has been destroyed. So instead of relying on the Python finaliser I added a release method which the windows can call during their destruction.

----

There is now two issues. There is the one about Cockpit's `Glyph.__del__` (the original issue reported here that happens in Linux and Mac) and there is the one about Freetype's `Face.__del__` (which seems to only happen on Mac).

But looking closer at the issues in the Mac, the issue about `Face.__del__` is not triggered by the fix for `Glyph.__del__`. That error was always there but it was mixed in the middle of the others and it's just too similar to the error from `Glyph.__del__`. So I believe we can push the fix for `Glyph.__del__` while trying to fix the other.

carandraug, 2021-03-10:
I pushed the fix for the deletion of `Glyph` (which I believe was about deleting gl textures after the gl context had been destroyed).

I'm still unsure about the freetype Face destruction which seems to happen on Mac only:

```
Exception ignored in: <function Face.__del__ at 0x7f7faa7c3160>
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/freetype/__init__.py", line 1140, in __del__
TypeError: 'NoneType' object is not callable
```

The `Face` class referred on the error message is the class from freetype-py and not the `Face` class from cockpit's freetype wrapper. I found a comment on rougier/freetype-py#44 of someone with the same issue on Mac. They may be related.

carandraug, 2023-05-18:
I've explored this in the freetype Python package (see discussion in rougier/freetype-py#169 and pull request in rougier/freetype-py#171) but the bottom line is that this is not our issue (but we sent a fix anyway). Closing.

closed 2023-05-18 by carandraug

iandobbie, 2024-01-11:
Just to note that new installs on macOS still need to use the master branch of freetype-py from github rather than the current release from PyPI to avoid this crash roughly 1 time in 5.