microscope-cockpit/cockpit #685: Can we please have step size changes from keyboard shortcuts back!

iandobbie, 2020-10-27:
The revised macrostage control has editable text boxes with step sizes in it but can we have a keyboard shortcut as well. Now pressing the old shortcut keys just gives a beep, I assume a rejected keypress. 

Navigating aorund the sample from the keyboard is now a real pain as you continually have to swap between keyboard and mouse to select the text box, then back to the keyboard to type the size and do the moves.

carandraug, 2020-10-27:
it works for me (Debian 10.6, Python 3.7.3, wxPython 4.1.0, cockpit 6d5aebc828540b10326 ).

1. start with `cockpit --no-config-files`
2. ensure that keyboard has num lock active
3. select cockpit main window
4. press 3 or 9 on the numpad changes the step size
5. select the macro stage window and repeat step 4

The above works for all windows except the python shell (because on the python shell window we want to actually type those characters).

The only important detail is that you need to select the window. If you're currently on a text box of the window, typing the character will try to type that character on that text box. Or maybe this is a Windows or Mac specific issue?

iandobbie, 2020-10-27:
I have been at home too long.

I was pressing 7 and 3. What planet am I on. Sorry please close the issue

Ian

closed 2020-10-27 by carandraug