microscope-cockpit/cockpit #689: SIM experiments fail half way through

NickHallONBI, 2020-11-06:
I am running a branch on my fork called "[dm_ians_edits](https://github.com/NickHallONBI/cockpit/tree/dm_ians_edits)". Essentially, this branch is the current master branch of cockpit plus the AO functionality which has not yet been added to the main repository.

When I run SIM experiments in this branch (2D or 3D), the experiment fails half way through. The following error is recorded in the "Standard error" section of the Logging panels window:

`Exception in thread Experiment-execute:`
`Traceback (most recent call last):`
`  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\socketutil.py", line 171, in receiveData`
`    chunk = sock.recv(min(60000, size - msglen))`
`socket.timeout: timed out`

`During handling of the above exception, another exception occurred:`

`Traceback (most recent call last):`
`  File "C:\Python37\lib\threading.py", line 917, in _bootstrap_inner`
`    self.run()`
`  File "C:\Python37\lib\threading.py", line 865, in run`
`    self._target(*self._args, **self._kwargs)`
`  File "c:\cockpit\nick\cockpit\cockpit\experiment\experiment.py", line 400, in execute`
`    curIndex + bestLen, numReps, repDuration)`
`  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 154, in executeAndWaitFor`
`    return executeAndWaitForOrTimeout(eventType, func, None, *args, **kwargs)`
`  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 188, in executeAndWaitForOrTimeout`
`    func(*args, **kwargs)`
`  File "c:\cockpit\nick\cockpit\cockpit\handlers\executor.py", line 206, in executeTable`
`    repDuration)`
`  File "c:\cockpit\nick\cockpit\cockpit\devices\executorDevices.py", line 202, in executeTable`
`    events.executeAndWaitFor(events.EXECUTOR_DONE % self.name, self.connection.RunActions)`
`  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 154, in executeAndWaitFor`
`    return executeAndWaitForOrTimeout(eventType, func, None, *args, **kwargs)`
`  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 188, in executeAndWaitForOrTimeout`
`    func(*args, **kwargs)`
`  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\core.py", line 185, in __call__`
`    return self.__send(self.__name, args, kwargs)`
`  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\core.py", line 453, in _pyroInvoke`
`    msg = message.Message.recv(self._pyroConnection, [message.MSG_RESULT], hmac_key=self._pyroHmacKey)`
`  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\message.py", line 168, in recv`
`    msg = cls.from_header(connection.recv(cls.header_size))`
`  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\socketutil.py", line 463, in recv`
`    return receiveData(self.sock, size)`
`  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\socketutil.py", line 184, in receiveData`
`    raise TimeoutError("receiving: timeout")`
`Pyro4.errors.TimeoutError: receiving: timeout`

The report in the 'Standard out' is:

`** WARNING **: file truncated - shape from header: (270.0, 512, 512). Expected to get 70778880 pixels but got 35389440 pixels`

Obviously the exact number of images/pixels expected and received changes with stack height but the failure point remains consistently at half way through the total expected frames/pixels. As far as I can tell, this issue only occurs for the SI experiments. Experiments such as the Z stack experiment run to completion.

I was previously running my '[Set_sensorless_params](https://github.com/NickHallONBI/cockpit/tree/Set_sensorless_params)' branch and this issue did not occur.

NickHallONBI, 2020-11-11:
This issue is actually more generalised than I first realised. It appears that failing half way through the experiment is also the case for Z stack experiments. The rotator sweep experiment appears to work fine though.

There are several further complications.

1) When the Z stack experiment fails, it doesn't produce the 'Standard out' report mentioned above, presumably because it does not attempt to reorder the stack. and therefore doesn't discover that half of the pixels are missing. This error is produced when the "View last file" functionality is utilised, presumably because at that point cockpit checks the actual number of pixels against the supposed pixels from the metadata and discovers half of them aren't there.

2) It is possible to run Z stacks which reach the half way point before the timeout error above is generated. In this case, the experiment still fails half way through, but the timeout error never occurs. 

3) Despite the rotator sweep experiment successfully completing, it still generates the timeout error:

'Exception in thread Experiment-execute:'
'Traceback (most recent call last):'
'  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\socketutil.py", line 171, in receiveData'
'    chunk = sock.recv(min(60000, size - msglen))'
'socket.timeout: timed out'

'During handling of the above exception, another exception occurred:'

'Traceback (most recent call last):'
'  File "C:\Python37\lib\threading.py", line 917, in _bootstrap_inner'
'    self.run()'
'  File "C:\Python37\lib\threading.py", line 865, in run'
'    self._target(*self._args, **self._kwargs)'
'  File "c:\cockpit\nick\cockpit\cockpit\experiment\experiment.py", line 400, in execute'
'    curIndex + bestLen, numReps, repDuration)'
'  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 154, in executeAndWaitFor'
'    return executeAndWaitForOrTimeout(eventType, func, None, *args, **kwargs)'
'  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 188, in executeAndWaitForOrTimeout'
'    func(*args, **kwargs)'
'  File "c:\cockpit\nick\cockpit\cockpit\handlers\executor.py", line 206, in executeTable'
'    repDuration)'
'  File "c:\cockpit\nick\cockpit\cockpit\devices\executorDevices.py", line 202, in executeTable'
'    events.executeAndWaitFor(events.EXECUTOR_DONE % self.name, self.connection.RunActions)'
'  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 154, in executeAndWaitFor'
'    return executeAndWaitForOrTimeout(eventType, func, None, *args, **kwargs)'
'  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 188, in executeAndWaitForOrTimeout'
'    func(*args, **kwargs)'
'  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\core.py", line 185, in __call__'
'    return self.__send(self.__name, args, kwargs)'
'  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\core.py", line 453, in _pyroInvoke'
'    msg = message.Message.recv(self._pyroConnection, [message.MSG_RESULT], hmac_key=self._pyroHmacKey)'
'  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\message.py", line 168, in recv
'    msg = cls.from_header(connection.recv(cls.header_size))'
'  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\socketutil.py", line 463, in recv
'    return receiveData(self.sock, size)'
'  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\socketutil.py", line 184, in receiveData
'    raise TimeoutError("receiving: timeout")'
'Pyro4.errors.TimeoutError: receiving: timeout'

This timeout error seems identical to the timeout error from the SI experiment before. The combination of points 2) and 3) leads me to believe that the timeout error and the experiments failing half way through are unrelated. If this is the case, then the situation is more complicated than previously though because it mean that there are no errors logged by cockpit or any of the remote device servers and yet the experiments just finish half way through.

NickHallONBI, 2020-11-11:
I have verified this functionality is consistent if I check out the MicronOxford master branch and run the same experiments. This bug does not seem to be introduced by the additional AO functionality changes/files in "dm_ians_edits" but rather resides in the master branch which that branch builds on top of.

iandobbie, 2020-11-13:
Nick and I have just found that it is a timing issue and the camera is triggered about 3ms too early. Still trying to find out why.

NickHallONBI, 2020-11-13:
Discovered the source of this error. The camera is missing every other trigger. This occurs because the spacing between triggers is such that, having sent one trigger, the subsequent trigger arrives at the camera while the camera is not armed and therefore cannot retrieve triggers. Line 669 of cockpit/experiment/experiment/py:

`nextUseTime += self.cameraToReadoutTime[camera] + decimal.Decimal(0.1)`

Determines this delay between triggers. Replacing this with

`nextUseTime += self.cameraToReadoutTime[camera] + decimal.Decimal(5)` 

Fixes the problem on DeepSIM, but it seems that the correct values should be fetched from the hardware, rather than being manually set by magic numbers in the code. Somewhere, we aren't fetching the correct time for how long the cameras are not armed and therefore not receiving triggers.

carandraug, 2020-11-13:
Any hint, at what might have caused the regression?

I'm looking at the changes in experiment since last October and I don't see anything that would cause this but found these two which roughly touch it:

* b3c58f244338c458ca7 removes a `camPseudoGlobalReadyTime` but that was unused
* fa3dbc4831baf80b9490833c72 removes the lights and cameras arguments from the experiment constructor

juliomateoslangerak, 2020-11-13:
I had similar issues and I went through the code to fix it. Also cause the triggering timing does not consider weather the camera has a global or rolling shutter which is a big issue with scmos cameras. I think this is a very important issue to fix and if you are going through the code now...
#665 and #651
You may have a look into my master branch. It is working for me now I refer to the commits in #651

NickHallONBI, 2020-11-13:
> Any hint, at what might have caused the regression?

I think I've found it. cockpit/devices/microscopeCamera.py, line 272-283 of Set_sensorless_params:

```
    def getTimeBetweenExposures(self, name, isExact=False):
        """Get the amount of time between exposures.

        This is the time that must pass after stopping one exposure
        before another can be started, in milliseconds."""
        # Camera uses time in s; cockpit uses ms.
        t = self.proxy.get_cycle_time() * 1000.0
        if isExact:
            result = decimal.Decimal(t)
        else:
            result = t
        return result
```

Same file, lines 254-268 in MicronOxford, master:

```
    def getTimeBetweenExposures(self, name, isExact=False):
        """Get the amount of time between exposures.

        This is the time that must pass after stopping one exposure
        before another can be started, in milliseconds."""
        # Camera uses time in s; cockpit uses ms.
        #Note cycle time is exposure+Readout!
        t_cyc = self._proxy.get_cycle_time() * 1000.0
        t_exp = self._proxy.get_exposure_time() * 1000.0
        t = t_cyc - t_exp
        if isExact:
            result = decimal.Decimal(t)
        else:
            result = t
        return result
```

The key difference is that in the working branch just the cycle time is used, whereas in the current master branch the exposure time is subtracted from the cycle time. In theory this is correct because the cycle time - exposure time should be the readout time, but in practice that proves to not be enough time between exposures. Adding an arbitrary additional delay of 5ms seems to fix this issue, but the real fix should be getting the correct readout time from the cameras.

It's worth noting that the cycle time scales with exposure time, so it is actually possible to get have a low enough exposure time (~3.5s) on the Set_sensorless_params branch such that the cycle time < the time when the camera isn't responding to triggers which leads to the same behaviour documented above. So before it was a corner case at very low exposure times, in the master branch it's a problem at all exposure times. All in all, it seems to be a more fundamental problem of getting the correct timing back from the camera, rather than an issue with cockpit.

carandraug, 2020-11-13:
When fixing this, consider that this change was introduced in ddc00f3aed09efae1be0d which fixed something else (not clear what, @iandobbie might remember better if there was a specific issue).

NickHallONBI, 2020-11-13:
Ian mentioned this when I pointed out where the change seemed to have come from. That commit was the right thing to do, as the time between exposures should be cycle time - exposure time and it wasn't previously. However, the time between exposures should be sufficient to clear the period when the camera isn't responding to triggers, which is clearly not the case. So the hack is just to add an additional 5ms to the time the camera returns, but the true fix will be to get the camera to return the correct time in the first place. 

Since the root issue seems to be with the cycle time being returned from the proxy (i.e. the microscope device) is incorrect, should we move this issue over to microscope? Or should it stay here?

iandobbie, 2020-11-13:
As Nick says the previous behaviour was wrong. We still have this on the Diamond CryoSIM system. If you set a 5s exposure, you get a 5s pause between frames which is totally unnecessary.

Another point to note is there is a way to do this correctly with the Andor cameras as I remember Mick and I chasing firmware updates as the iXons used to have a variable delay after a trigger before starting capture, requiring an additions 10 ms or so to be added to this timing to make it work reliably. We upgraded the firmware on the camera and this went away. Do note this was in pre-microscope days so when the camera driver was a separate piece of code. I'm sure it will still be around in Mick's github repository so maybe we should investigate there how it was solved last time?

carandraug, 2020-11-14:
> Since the root issue seems to be with the cycle time being returned from the proxy (i.e. the microscope device) is incorrect, should we move this issue over to microscope? Or should it stay here?

I think this issue can stay here, it has a lot of debugging unique to cockpit.  But do open an issue in the microscope repo with the code to reproduce using microscope only code.

iandobbie, 2021-04-26:
The cycle time issue has been fixed in the atmcd.py andor emccd driver. It was using the exposure time +readout time and assiming this was the cycle time but in some modes (including the one used) there are also keep clear cycles to be considered. You just need to use a different value returned by the camera DLL. Closing as fixed in microscope.

closed 2021-04-26 by iandobbie