microscope-cockpit/cockpit #690: Single snaps too quickly can  crash the DSP

iandobbie, 2020-11-11:
The real DSP cannot cope with single snaps very quickly, repeated pressing of either the snap button or the snap key (numpad 0) will repeatably crash the DSP.

There is code to fix this in the takeImage function, however it lacks two important factors to enable it, a) the takeImage function needs to be called with shouldBlock=True. Why isnt this happening on snaps? b) there needs to be a sensible number in the camLimiter threshold in the getNextImageTime. 

At Diamond we set both the GUI and keyboard calls to takeImage to add the shouldBlock=True option and then set the camLimiter to 500 (0.5s, it is in ms). This is probably too conservative but we will see if it prevents the crashes we were seeing.

juliomateoslangerak, 2020-11-13:
Funny. I have a similar issue with the FPGA but very random.
Happens also when I use the live mode.

iandobbie, 2020-11-24:
Have you tried setting the shouldblock=True option and the camLimiter variable which should force a delay between snaps.