microscope-cockpit/cockpit #692: mosaic fails with latest wxPython (4.1.1)

carandraug, 2020-12-06:
The mosaic window does not work with the latest wxPython release (4.1.1). When starting cockpit, we get the following on stderr:

```
Traceback (most recent call last):
  File "/home/carandraug/.local/lib/python3.7/site-packages/OpenGL/latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/gui/mosaic/canvas.py", line 351, in onPaint
    self.initGL()
  File "/home/carandraug/src/cockpit/cockpit/gui/mosaic/canvas.py", line 177, in initGL
    self.megaTiles.append(MegaTile((-x, y)))
  File "/home/carandraug/src/cockpit/cockpit/gui/mosaic/tile.py", line 302, in __init__
    megaTileFramebuffer = glGenFramebuffers(1)
  File "/home/carandraug/.local/lib/python3.7/site-packages/OpenGL/latebind.py", line 46, in __call__
    self._finalCall = self.finalise()
  File "/home/carandraug/.local/lib/python3.7/site-packages/OpenGL/extensions.py", line 245, in finalise
    self.__name__,
OpenGL.error.NullFunctionError: Attempt to call an undefined alternate function (glGenFramebuffers, glGenFramebuffersEXT), check for bool(glGenFramebuffers) before calling
```

and the following on stdout:

```
Error rendering the canvas: Attempt to call an undefined alternate function (glGenFramebuffers, glGenFramebuffersEXT), check for bool(glGenFramebuffers) before calling
```

This happens on Linux (GTK3), not sure if the issue is platform specific. The issue is not present on wxPython 4.1.0, only on wxPython 4.1.1. But the errors come from OpenGL.

iandobbie, 2020-12-07:
This does not seem to be a problem on macOS. I just ran with wxpython 4.1.1 and everything I tried worked. However the openGL components still have an issue on the mac retina displays that they only display in a quarter of the expected panel (ie half size in x and y)

carandraug, 2021-03-09:
On `cockpit.gui.mosaic.tile` we are wildcard importing `OpenGL.GL` and `OpenGL.GL.framebufferobjects`. The later import overloads the `glBindFramebuffer`, `glFramebufferTexture2D`, and `glGenFramebuffers` functions with something that makes use of the GL_EXT_framebuffer_object extension. If I remove the overload imports, I no longer get an error. I guess that the import of `OpenGL.GL.framebufferobjects` is done so that we can support older OpenGL versions. Frame buffer objects were added to core in version 3.0 which I think is safe enough.

While removing the use of the FBO extensions removes the error, the behaviour is still broken. The mosaic canvas part of the mosaic window is shrunk to 1 pixel wide. I'm sure what is going on, when the window is show it gets two size events and the second one is shrinking the canvas. Passing an initial size to the canvas, instead of having the canvas expand to the window client size, fixes the issue.

I think part of the issue is the naked mosaic frame. We should have a simple Frame with only the panel inside. But there's some hacks there that require it to be a Frame. We should probably also split the MosaicWindow into panels with individual functionality.

Anyway, not very happy about not understanding the root of this issue but seems fixed. I've also tested it on mac and Windows. Closing as fixed.

closed 2021-03-09 by carandraug