microscope-cockpit/cockpit #694: Cockpit won't start on Mac using Conda Environment

beniroquai, 2021-01-01:
After installing the toolbox using 

```
pip install microscope-cockpit
```

or using ```git``` and doing a ```pip install -e ./```

I receive the following message in a conda environment:

```
bene$ cockpit
This program needs access to the screen. Please run with a
Framework build of python, and only when you are logged in
on the main display of your Mac.
```

Doing the same on a plain python 3.8 installation gives the following error when starting cockpit from the locally installed version:
```py
bene$ python3 gui/__init__.py 
Traceback (most recent call last):
  File "__init__.py", line 28, in <module>
    import cockpit.events
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/cockpit/cockpit/__init__.py", line 80, in <module>
    import cockpit.gui.mainWindow
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/cockpit/cockpit/gui/mainWindow.py", line 72, in <module>
    import cockpit.gui.fileViewerWindow
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/cockpit/cockpit/gui/fileViewerWindow.py", line 53, in <module>
    from cockpit.gui import imageSequenceViewer
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/cockpit/cockpit/gui/imageSequenceViewer.py", line 54, in <module>
    import cockpit.gui.imageViewer.viewCanvas
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/cockpit/cockpit/gui/imageViewer/viewCanvas.py", line 57, in <module>
    import cockpit.gui.freetype
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/cockpit/cockpit/gui/freetype.py", line 55, in <module>
    import freetype
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/cockpit/cockpit/gui/freetype.py", line 113, in <module>
    class _Glyph:
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/cockpit/cockpit/gui/freetype.py", line 114, in _Glyph
    def __init__(self, face: freetype.Face, char: str) -> None:
AttributeError: partially initialized module 'freetype' has no attribute 'Face' (most likely due to a circular import)
```

Any idea what could be wrong here?

beniroquai, 2021-01-01:
Sorry, found the error. Had to use ```pythonw``` instead of ```python```:

[image]

Also the I had to do ```pythonw __init__.py``` in the root folder of course..

iandobbie, 2021-01-01:
Hi.

Glad you got it sorted but the macOS python is a real pain. We suggest you just install the python.org version which solves most of the issues. Alternatively you can use homebrew to install python which is what I do.

If you then run the setuptools installer you get a runable program called cockpit so you can connect this to an icon or run from the terminal (which I do on my Mac laptop)

I have heard that apple are planing to remove python from the system install which should solve this.

Ian

[image]<https://user-images.githubusercontent.com/4345528/103445582-60463d80-4c76-11eb-93a0-64551bd7976d.png>

carandraug, 2021-01-04:
Hi Ian, when I last spoke with Benedict (@beniroquai), he mentioned that he uses anaconda's distribution of Python and not the macOS python distribution.

@beniroquai : I'm not a Mac but from what I could find online, your solution of using `pythonw path-to-program` is the recommended solution and it's unlikely that the issue will be fixed (see [stackoverflow post](https://stackoverflow.com/questions/44549306/how-to-make-an-existing-wxpython-code-run-on-mac-with-conda)).

By the way, instead of specifying the path to the cockpit program, I have now pushed a new `__main__.py` module so you can do:

    pythonw -m cockpit

previously you'd have to know it's install path or use `cocpkit.__init__`

Anyway, I'm closing this as invalid (expected behaviour - won't be fixed), since there's nothing we can do about it (maybe add a note on the install documentation?).

closed 2021-01-04 by carandraug