microscope-cockpit/cockpit #695: Camera interface with python-microscope is broken

dstoychev, 2021-01-14:
Commit python-microscope/microscope@53412c7f2216654d166101d19772f520e5f3eff changes the camera enabling interface. Cockpit cannot rely on the return value of the `enable()` method any longer. My quick fix is to query the enable status with the `get_is_enabled()` method, e.g. fea32089af70257b678f45b4f1c5f13e9a6d5432.

carandraug, 2021-01-14:
The commit you reference in Microscope does not exist but I think you meant python-microscope/microscope@853412c7f2216654d1 so I edited your comment.

Some context, the `enable` method could raise an issue if enabling failed, or it could return `False`. This meant you had to both check the return value and guard against an exception. That commit changes it so that `enable()` returns `None` and one just needs to guard against an exception (in theory it seems possible that `enable` can fail and not raise an error if `_do_enable` returns `False` but in practice I'm not sure if that ever happens).

carandraug, 2021-01-14:
Anyway, I don't think your quick fix on fea32089af70257b678f45b4f1c5f13e9a6d5432 quite works. Because that's async code, you may be checking the result of `get_is_enabled` before `enable` had time to finish. You need to wait for `enable` to return and only then call `get_is_enabled` (which no longer needs the async proxy).

carandraug, 2021-01-14:
> Because that's async code, you may be checking the result of `get_is_enabled` before `enable` had time to finish. 

I confirmed that this could happen, just get a simulated camera whose enable simply sleeps for a couple of seconds.

I ammended your commit into 89024ff022ccd3b75f163b857d18fde3bfda0fc7and pushed.

closed 2021-01-14 by carandraug