microscope-cockpit/cockpit #700: touchscreen code crashes if a light doesn't have an associated wavelength.

iandobbie, 2021-02-04:


iandobbie, 2022-05-26:
Fixed by setting to -50 if not defined, gives gray colour bar [f0a5486](https://github.com/MicronOxford/cockpit/commit/f0a548630e1e3fbc718230e91d6c2b90d3fc66de)

carandraug, 2022-05-30:
I really don't think this is an appropriate fix, for two reasons:

1. Not having a wavelength means no wavelength, it does not mean `-50nm`. Having a `-50` as special case is just weird when `None` is a just as good special case with much more obvious meaning.
2. The gui is changing property of the handle (which is shared across the rest of cockpit), if such a thing is done, then it should happen when the handler constructor. The wavelength should be a read only property of the handler.

carandraug, 2022-05-30:
Interestingly, if instead of trying to get the wavelength from the `LightPowerHandler` you use `LightHandler` (use `self.light` instead of `self.power`) then it works because the latter handles the special case of `None` internally by setting it to zero. I never quite understood why the separation of those two handlers so I'm not sure which one makes more sense.

closed 2022-06-28 by iandobbie