microscope-cockpit/cockpit #704: Clone/pull failed because of invalid path

dstoychev, 2021-02-05:
This may be Windows related, but pulling or cloning fails at the moment. Sample output:

```
$ git clone https://github.com/MicronOxford/cockpit.git
Cloning into 'cockpit'...
remote: Enumerating objects: 195, done.
remote: Counting objects: 100% (195/195), done.
remote: Compressing objects: 100% (133/133), done.
remote: Total 12435 (delta 89), reused 152 (delta 61), pack-reused 12240
Receiving objects: 100% (12435/12435), 14.55 MiB | 742.00 KiB/s, done.
Resolving deltas: 100% (9359/9359), done.
error: invalid path 'aux/convert-images-to-png.py'
fatal: unable to checkout working tree
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'
```

The same `error: invalid path 'aux/convert-images-to-png.py'` is given when I try to pull on a different (Windows) computer.

dstoychev, 2021-02-05:
Seems like "aux" is a reserved filename under NTFS. It'd be nice to rename the new aux directory. The alternative is for users to set the `core.protectNTFS` Git flag to false.

iandobbie, 2021-02-05:
Yes but having a file called aux on windows is a bad idea, like COM and PRT. Oh the joys of historical baggage. Just change the dir name.

carandraug, 2021-02-05:
I was not aware of AUX being a reserved filename. It's just a name I've seen before to store the build auxiliary tools. So I renamed to tools. Closing as fixed.

closed 2021-02-05 by carandraug

mickp, 2021-02-05:
Full list of prohibited names for reference, from https://docs.microsoft.com/en-us/azure/devops/organizations/settings/naming-restrictions

[image]

iandobbie, 2021-02-08:
Reading deeper the windows file name space becomes even more arcane. Check out the  "\\?\" prefix, you can then insert "/./" and "/../" into file paths. The joys!