microscope-cockpit/cockpit #705: Colour in the camera views.

iandobbie, 2021-02-08:
I have added the option to display a individual camera views in red, green or blue by adding an additional option to the shader which just renders the intensity as red green or blue. A relatively simple change that should have no affect on performance. I think we should just pull it, any objections?

https://github.com/iandobbie/cockpit/tree/colour-views

carandraug, 2021-02-11:
The proposed change adds support to to colorise in red, green, or blue via a menu entry for each colour in each view.

This solution has the issue that will not work together with the show clip option.

Also, I feel that this solution will not extend nicely to other colours, not only on the UI (a menu entry per colour) but also on the code (every colour gets a number -- and if we do get a number for each colour we should at least use enums).

Having to manually pick the right colour for each view is not ideal. Not only because one can pick the wrong colour, but also because the right colour to use can change if we change the filterwheel. I think this can be avoided since we know the channel colour (we use it to colour the top bar of the camera view). If so, we could have a single "Colorise" menu entry which picks the right colour and uses it.

iandobbie, 2021-02-11:
I don't think it clashes with the show clip entry. The two are mutually exclusive, show clip basically doesn't work on coloured images as you might not see what is clipped due to the colour choice.  I realise that my gray option did break this.  So we need a better selection for this. 

I don't agree that we should just set the real colour a leave it at that. What happens if you are imaging at 620 and 680 both will be red? Or 360 and 460, both will be blue? Passing a number of some sort to the shader is more sensible than having code in the shader to interpret enums, strings or other input. We want the shader to be as simple as possible.

My suggestion: We automatically select colour but also allow it to be changed. We should probably move the colour selection to a sub menu. I would start with RGB, gray, show clip and we can add more LUTs if people really fancy them.

carandraug, 2021-02-11:
> I don't think it clashes with the show clip entry. The two are mutually exclusive, show clip basically doesn't work on coloured images as you might not see what is clipped due to the colour choice.

That's my point. They shouldn't be mutually exclusive. Why would users not be interested in seeing over/under saturation just because they are also interested on seeing the image with the correct colour? This would require adjusting the clipping colours based on the colour used to display.

> I don't agree that we should just set the real colour a leave it at that. What happens if you are imaging at 620 and 680 both will be red? Or 360 and 460, both will be blue?

You are right. But at least they will be displayed with the real (excitation) colour. My understanding is that when people want to colourise the output, they want to see the "real" colour. This means somehow mapping the image itself to the excitation wavelength and not map a camera to a single colour.

> Passing a number of some sort to the shader is more sensible than having code in the shader to interpret enums, strings or other input. We want the shader to be as simple as possible.

You can just pass int enums and still have an equally simple shader. But having false colour in Python being an int is weird. Why would `self.image.falseCol(3)` mean set colour to blue instead of some other colour? I see the 123 colour being RGB but that will also not scale. It's also easy to see that at some point we will want to have more colours so I'm not keen on merging a solution that already has that limitation.

> My suggestion: We automatically select colour but also allow it to be changed. We should probably move the colour selection to a sub menu. I would start with RGB, gray, show clip and we can add more LUTs if people really fancy them.

A sub menu for the colouring might be better. But needs to be clear what colour to use on the different cases. This is not only map camera view to a single colour.

iandobbie, 2025-09-27:
I have been thinking about colour camera views as well as colour mosaics. The obvious answer for arbitary colour is to pass an rgb tuple and raw intensity data and have the shader multiple the intensity by the rgb tuple to get the ouput rgb value. This would give variable intensity at virtually any colour and give a "real" colour image for many configs.