microscope-cockpit/cockpit #706: Camera transform is not properly initialised

dstoychev, 2021-02-08:
Two things are missing:
1) The initialisation of the `_transform` attribute of `CameraDevice` classes. It makes sense to me that it's initialised to the `baseTransform` attribute, which is just the "transform" key of the section in the config file.
2) Microscope needs to be informed about the initial transform value. Probably best done at the end of `MicroscopeCamera.initialize()` method. It should call the `set_transform()` proxy method and pass the base transform to it (or the `_transform` attribute if it was set to the base, as suggested in point 1 above).

Example changes that address both of these points:
```python
class CameraDevice(device.Device):
    def __init__(self, name, config):
        super().__init__(name, config)
        # baseTransform depends on camera orientation and is constant.
        self.baseTransform = Transform(config.get('transform', None))
        self._transform = self.baseTransform
```
```python
    def initialize(self):
        # Parent class will connect to proxy
        super().initialize()
        # Lister to receive data
        self.listener = cockpit.util.listener.Listener(self._proxy,
                                               lambda *args: self.receiveData(*args))
        try:
            self.updateSettings()
        except:
            pass
        if 'readout mode' in self.settings:
            self.modes = self.describe_setting('readout mode')['values']
        else:
            self.modes = []
        self._proxy.set_transform(self._transform)
```

iandobbie, 2021-03-15:
I suggest an alternative fix in  706-set-camera-transform, which uses the existing _setTransform call and will updated the remote settings dict as well.

dstoychev, 2021-03-16:
Yes, seems like a much better solution. I tested it today and it works.

iandobbie, 2021-03-19:
Ok, pushed to master and closing as fixed

closed 2021-03-19 by iandobbie