microscope-cockpit/cockpit #707: Broken config files can lead to no feedback failure to start.

iandobbie, 2021-02-09:
I have cockpit installed as a clickable icon on the desktop. I managed to create a Zpiezo with 2 "gain: " lines. This caused a crash in the startup code but absolutely no feedback to the user. I had to open a shell and start cockpit from there to fin out what the issue was.

carandraug, 2021-02-09:
Configuration is read before starting the actual wx app. I believe the root issue is the same as the root issue of #439 .

carandraug, 2021-02-11:
I have pushed the fixed proposed in #439 (see 2f4f58d6defcfe29e368ace29cb1e7e819ff715d). Errors happening during the "pre" initialization steps will now be displayed on a GUI. If the failure is because it even fails to import the modules, that is not caught though.

closed 2021-02-11 by carandraug