microscope-cockpit/cockpit #708: Experiment issues on Oxford cryosim setup.

iandobbie, 2021-02-10:
I have some very weird experiment issues on the Oxford CryoSIM setup. 

1) all hardware is connected and hw triggers appear to work including SLM, lasers and camera.

2) Camera is set in HW trigger mode

3) Snaps and mosaic images work fine, triggering both camera and laser

4) experiments appear to be created correctly, but the system is stuck in waiting for image 0/X. This happens for both plain Z stacks and SIM experiments. 

Any debugging advise?

carandraug, 2021-02-10:
Which camera is on the system?

iandobbie, 2021-02-10:
Its an andor ixon.

The symptoms are even weirder than I thought. 

The live button doesn't work on startup, but with some fiddling it starts to work. Maybe involving going to ambient light source. Not exactly tracked down what makes it work. 

Oscilloscope shows that hw trigger to laser and camera are working and being triggered. Action table seems reasonable but it just isnt running. 

Z piezo signals seem to be working. 

Next thing I am going to try is rebooting the redpitaya, maybe even swap it out I have one for  the zaber here. Unfortunately its boxed up!.

iandobbie, 2021-02-10:
I merged in the latest master cockpit version. It has Nick's DM code so it it is not just master. But I might try reverting to plain master too

carandraug, 2021-02-10:
May be better to check the exact commit for microscope (both local and in the camera device server) and cockpit. Currently, the latest master cockpit does not have Nick's DM code, so I'm not sure what exactly are you running.

iandobbie, 2021-02-10:
Oh I am running latest cockpit with Nicks stuff on top. 

Further info. experiment runs fine. laser triggers and the Z piezo ramps as expected but the system is not sending any camera triggers. All very strange. 

I also just updated the microscope remote as I thought that the relatively recent trigger changes might be the cause. I am going to run it under pycharm once the lab meeting is over.

iandobbie, 2021-02-10:
Something is dropping the camera trigger. Here is the first few lines of action table in cockpit....

experiment.lastExperiment.table
    0.00             2 zPiezo    60.32477613223105
    0.00                  488                 True
    0.00           East-Green                 True
    0.10           East-Green                False
  100.00                  488                False
  100.00             2 zPiezo    60.32477613223105
  100.05             2 zPiezo    61.82477613223105

And from the redpitaya

0 4 1
0 -1 2246.0
100000000 4 0
100050000 -1 2301.8477

The camera trigger just inst there. I wonder is soemthig is assuming 8 digital lines as the camera is pin 9, whereas the laser is pin 4.

iandobbie, 2021-02-10:
Yes moved the trigger to pin 2 and things worked.

iandobbie, 2021-02-10:
Heart of the issue in #709

closed 2021-02-10 by iandobbie