microscope-cockpit/cockpit #710: Non-existant depot file doesn't generate an error.

iandobbie, 2021-02-12:
If you start cockpit with `cockpit --depot-file missing-file.depot` the fact the specified depot file is missing is ignored and the default behaviour (maybe a simulated system) occurs. Although no technically a bug it does lead to unexpected behaviour. I think a specified but missing file should generate a warning or even a failure to start.

carandraug, 2021-02-12:
That's a good point. I think it should error too.

iandobbie, 2024-03-20:
Above commit raises a generic exception, maybe we should have a different exception, and I have no idea about other missing config files.

iandobbie, 2024-03-20:
Unfortunately this fix breaks the default of no config files at all, and says it cant read the default location depot.conf. I guess I need to think a little harder about how to implement this check.

carandraug, 2024-03-20:
The reason this is happening boils down to a design choice of configparser (from `ConfigParser.read`):

> Files that cannot be opened are silently ignored; this is designed so that you can specify an iterable of potential configuration file locations (e.g. current directory, user's home directory, systemwide directory), and all existing configuration files in the iterable will be read.

If we change this, we should also change the behaviour in `CockpitConfig` for consistency.

But maybe we could check if the filepaths exist when we parse the command line options and config file, check if the file exists at that point, and warn about it then?

iandobbie, 2024-03-20:
Yes, I am just trying to think of the best way to deal with this. I think the existing behavIour  with the default locations is useful to read system and user default files if none are specified.  

As you say I think the best approach is to test if a specified command line option is readable at parse, and warn or die then. As you say we should also do the same for config files as knowing that you typo'd the command line is useful.

iandobbie, 2024-03-21:
I have implemented this in the commit above. My only question is should we raise some specific exception or is it ok to just raise the generic Exception()? I thought about a modal dialog but I don't think our gui is running yet. 

By checking the command line argument explicitly is doesn't brake the behaviour of seraching for system and user specific files when no command line arguments are given.