microscope-cockpit/cockpit #711: Slight issue with action plotter.

iandobbie, 2021-02-12:
The action plotter does not deal well with digital values. 
Say we have a signal like the following
Time   state
0          False
10        True
10.1      False
20        True

This will generate a sawtooth diagram rather than the expect set of delta functions. I understand that analogue values should probably be plotted like this but digital ones need to be in their previous state immediate before the state change as this is how that are in real life.

iandobbie, 2021-02-12:
The following code doesn't work.

     if isinstance(action, (float, int)):
            return "analogue"

from a shell.... 
>>> isinstance(True, (float,int))
True

iandobbie, 2021-02-12:
Fixed this by reversing order of test. Also added code to add an additional data point at this time and with the previous digital state, which fixes the sawtooth rather than square/delta function images.

closed 2021-02-12 by iandobbie

carandraug, 2021-02-12:
This is because in Python `bool` subclasses from `int`:

```
>>> isinstance(True, int)
True
>>> issubclass(bool, int)
True
```

@iandobbie please next time add a reference to the issue number on the commit message. Also, please fix one thing per commit.