microscope-cockpit/cockpit #713: Startup pause to allow microscope to start.

iandobbie, 2021-02-24:
If microscope is started immediately before cockpit, then cockpit can try to connect before microscope is running. Additionally in the simulation mode this startup script has a delay but there is no visual feedback during this delay.

I suggest we put a poll loop around the first microscope connection to allow a pause in the most common situations where all devices are in one microscope process.  I have a suggested fix in https://github.com/iandobbie/cockpit/tree/startupPause

carandraug, 2021-02-25:
The proposed fix means that if connection to the device fails, the user does not get any information about which device failed.  The only message received is "Failed to start the device server".

20 seconds timeout also seems excessive, specially since there's seem to be no method to cancel the startup.

Also, I would contest the premise that the most common situation is where all devices are in one microscope process, I think that only the simulation has all the devices in the same process. If the reason to add this is to cover the situation when cockpit is started immediately after starting one of the device servers, I see no reason why that device server could not be the second or the last device that cockpit tries to connect to.

> Additionally in the simulation mode this startup script has a delay but there is no visual feedback during this delay.

But that simulation could have a visual feedback. It might even be better if it does. The simulation "mode" starts the device server and then starts cockpit. The proposal here is to have cockpit do the wait and print error message. But cockpit is not in a position to give back useful error messages if the device server failed to start. Instead, the simulation "mode" should start the device server and wait until that works before starting cockpit. The simulation "mode" could then get errors from the device server and present a useful error message.

iandobbie, 2021-03-19:
Ok, case accepted. I guess I need to open an issue on the cockpit-distribution to add a gui window to show something is happening in the simulation mode startup

closed 2021-03-19 by iandobbie