microscope-cockpit/cockpit #715: Closing cockpit leads to a segv about 90% of time on a Mac.

iandobbie, 2021-02-25:
Looking at a few of the crash dumps I have shows it seems to be a:   
wxEvtHandler::SafelyProcessEvent(wxEvent&) 
call that kills things. I assume that wx events are still hanging around after things they refer to have been deleted so they cause access to illegal memory.

Reading a few threads online seems to suggest this is a common problem with wx on  a Mac with events unhandled. They suggest using a wx.CallAfter() or wx.CallLater() call to close the program at the end to allow these events to be handled.

My issue is I cant see where the close event actually happens. Is it the mainWindow.py OnClose event or elsewhere? How are the child threads cleaned up etc...

carandraug, 2021-02-25:
Where threads are closed depends no where the thread has been started.  `CockpitApp.OnExit` is likely to be the thing that triggers  be where most stuff happens. If there's any threads after `Cockpit.OnExit`, then there's a piece of code in `cockpit.main` that checks for any remaining threads and closes them. It also logs information about those threads. Do you see anything in the cockpit log files?

iandobbie, 2021-02-26:
A segement of a typical crash dump

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [46280]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libwx_baseu-3.1.5.0.0.dylib         0x000000010a0ce249 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 9
1   libwx_baseu-3.1.5.0.0.dylib         0x000000010a06108a wxTimerImpl::SendEvent() + 90
2   _core.cpython-39-darwin.so          0x0000000108de7d69 sipwxTimer::Notify() 

[Python_2021-02-26-104324_Ians-MacBook-Pro-7.crash.txt](https://github.com/MicronOxford/cockpit/files/6049134/Python_2021-02-26-104324_Ians-MacBook-Pro-7.crash.txt)

carandraug, 2021-03-10:
I've been testing Cockpit on a Mac and I have seen it crash at exit in a few different ways. The error message is never much detailed and neither are the crash reports. We're probably not cleaning up things properly but I guess we're doing it wrong in more than one place.

iandobbie, 2021-06-16:
Tom has a suggested fix in https://github.com/thomasmfish/cockpit/tree/potential-exit-crash-fix needs to be investigated

thomasmfish, 2021-06-16:
The crash happening on the B24 SIM PC (Windows) wasn't helped by my changes, so probably not worth considering

iandobbie, 2022-12-21:
Just to update this thread. I am not sure exactly what changes have imporved things but it now happens much less frequently. Maybe 10% of time.

iandobbie, 2022-12-22:
Just for reference a crash I received today was a segv due to a wxtimer, an exert of the crashdump is:

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libwx_baseu-3.2.0.0.0.dylib   	       0x1056fb464 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 16
1   libwx_baseu-3.2.0.0.0.dylib   	       0x1056973e8 wxTimerImpl::SendEvent() + 100
2   libwx_baseu-3.2.0.0.0.dylib   	       0x1056973e8 wxTimerImpl::SendEvent() + 100
3   _core.cpython-310-darwin.so   	       0x1070f4ed4 sipwxTimer::Notify() + 128
4   CoreFoundation                	       0x18f7eb8cc __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 32

So I still think we are not properly closing everything on shutdown.

iandobbie, 2023-06-20:
Ok, so I now have a branch that deals with the safecontrol timers on the lights panel. it propagates the timer objects up the call stack so they are available as a list in the mainWindow and then loops through the list and stop each timer in the onClose function.

This is in my branch https://github.com/iandobbie/cockpit/tree/timer-crash-fix .

Advantages: relatively minor changes in gui/mainPanels and gui/mainWindow 

Disadvantages: relies on good housekeeping from any functions that use the relevant safecontrols which set a timer to ensure that the timer is added to the self.timers list in mainWindow.

I think this is a reasonable solution, but needs code added to the other items which use safecontrols with timers, which appears to be a single control in the touchscreen, where a similar fix could be applied. 

This also does not yet deal with the timer using futures.ThreadPoolExecutor() in the light power handler that also seemed to be causing at least some crashes on my Mac. I will check if this is still an issue.

iandobbie, 2023-06-20:
Ok, more updates, 10 shutdowns... 3 crash with freetype issue, 3 crash with wxtimer issue. 

I think this is better but obviously still needs work.

iandobbie, 2023-06-20:
Ok, did some more searching and the IconButton function on the touchscreen starts a wx.Timer for each button, no wonder we were crashing often, this is a further 29 timers. Some unpleasant hacking has got all these in a list which I am also then calling .Stop() on in the ONQuit touchscreen function as onclose is remapped to iconise I believe.

Now 10 runs got me 2 freetype crashes and one from the light-power handler _updater that still isnt dealt with as I don't quite see how to  propagate its timer object. The code starts like this....

    ## We use a class method to monitor output power by querying hardware.
    # A list of instances. Light persist until exit, so don't need weakrefs.
    _instances = []
    @classmethod
    @cockpit.util.threads.callInNewThread
    def _updater(cls):
        ## Monitor output power and tell controls to update their display.
        # Querying power status can block while I/O is pending, so we use a
        # threadpool.
        # A map of lights to queries.
        queries = {}
        with futures.ThreadPoolExecutor() as executor:
            while True:

Any suggestions on how to get at the executor and issue an executor.shutdown() to kill the process and stop the crashes.

iandobbie, 2023-06-20:
Even with the code currently in my branch there is still an occasional wx.timer related crash so I am obviously missing some. I'm not sure if it is a safe control related one or if I have missed some of the iconButtons in the touchscreen

iandobbie, 2023-06-21:
Still chasing my tail here. I started to print every time a timer is created then looked at my list before killing them. Created 38 deleted 29, so I am missing a bunch of timers, I think in the touchscreen IconButton code but I will have to chase this down.

iandobbie, 2023-06-21:
Adding a 1 second delay after the wx.Timer.Stop() calls didn't solve the occasional crash issues and noticeably slowed down the exit process.  Not sure what the next step is, however the frequency of issues is much reduced.

The touchscreen code definitely needs cleanup and we should also survey the aurox and boulder devices as they likely have a similar issue. 

As an aside I also updated my freetype library from the github repository and that crash has vanished as well.

carandraug, 2023-06-21:
I really don't think that the parent should be collecting and stopping the timers from the child windows. Each window should clean after themselves, it's their responsibility.

1. Having things in separate panels and frames enables reuse of the elements elsewhere without having to worry about its internals. That change goes against that principle. It will be more difficult in the future if we can't trust that each panel is self-contained.

2. we currently have this thing where windows are never closed, they're just hidden. Would be really nice if we could really close them. Having the main window responsible from stopping timers of all children windows, it's another step in the other direction.

iandobbie, 2023-06-21:
As far as I can tell objects like a toggleButton, which some of these classes are derived from don't get an EVT_CLOSE signal as I did try this approach as well. I'll do some more research.

iandobbie, 2023-06-21:
Ok, I'll try that instead. Much less work than keeping track of the timers up the call stack.

iandobbie, 2023-06-21:
The light power handler updater is a lot more complicated to solve. The futures.ThreadpoolExecutor is not easy to shutdown as this generates many more errors. I have also tried to add a variable to the while loop to cause termination but this doesn't seem to work either.

I am going to push the minor edits to safecontrols and touchscreen.

iandobbie, 2023-06-21:
Ok, the wx.timers code is in commit d10bbe1 and the freetype errors goes away with a development branch of freetype so it is now only the very occasional error from the lightpower handler updater and I can't work out how to properly kill that.

carandraug, 2023-06-21:
> Ok, the wx.timers code is in commit https://github.com/microscope-cockpit/cockpit/commit/d10bbe1d8bcdf5b818a9453176bfc072fe129e9d

If you can mention the issue number in the commit message, e.g. `Stop wx.timers on shutdown in safecontrols and touchscreen (#715)`, then we can link the commit back here. The idea is that, typically, one finds the commit when searching the logs for why a change was done. At that point, it is useful to have a link back to an issue discussion. Without that reference on the commit message, it's tricky to find the related discussion.

iandobbie, 2023-06-23:
Hi David, good point. I will try to remember in future. Unfortunately I think editing history is likely to break things for people as you have done some changes ontop of my commit.

I am going to close this issue as 95% of the issue is solved and the remaining _updater issue in the lightpower handler is specific and identified. The new issue is #865 .