microscope-cockpit/cockpit #716: option to clear user config / cache

carandraug, 2021-03-03:
Other than cockpit's config (`cockpit.config`), there is config.py (`cockpit.util.userConfig`) which is more like user cache. This is undocumented, and mostly remembers last used values. Ilan requested that there was a way to clear this file in case users entered something weird.

carandraug, 2021-03-03:
I think we could have a menu entry with something like "Restore Defaults" which would simply empty the file since the defaults are in code when reading a value that does not exist.

carandraug, 2021-03-03:
Proposed fix on my "716-restore-defaults" branch. It adds an "Edit>Reset User Configuration" menu entry. This does not reset the window positions for two reasons:

1. the default position of a window is OS preferences specific and will often be dependent on the windows open at the time.
2. when we exit cockpit, cokcpit would save those positions so they would be kept for the next cockpit session

But I don't think that window position was the problem anyway. Selecting the window on the windows menu should move them to the monitor if they somehow are outside the view.

iandobbie, 2021-03-04:
> But I don't think that window position was the problem anyway. Selecting the window on the windows menu should move them to the monitor if they somehow are outside the view.

Is this a suggestion of new functionality, or your expectation of how it works. This is definitely not how it works on the mac at the moment. 
Connect a second monitor
Move cockpit windows onto the second display
Close cockpit (may require a clean close that doesnt often happen)
Disconnect second monitor
restart cockpit

you will have windows that you cannot interact with. As we no longer have the move-to-mouse option you can only get them back by editing or removing the config file. Unfortunately your removal option doesnt fix this as when cockpit exits the window positions will be saved again.

carandraug, 2021-03-05:
> Is this a suggestion of new functionality, or your expectation of how it works. This is definitely not how it works on the mac at the moment. [...]

This is how it should be working.

> [...] As we no longer have the move-to-mouse option you can only get them back by editing or removing the config file. [...]

We have the move to mouse option, this was discussed on #652 . The point of move to mouse was to move windows that somehow ended outside the monitor. When you select the window from the menu, it's check if it's in a valid position and if so, it moves it to the mouse position. The code in question is in `cockpit.gui.mainWindow.WindowsMenu.OnWindowTitle`:

        if wx.Display.GetFromWindow(window) == wx.NOT_FOUND:
            window.SetPosition(wx.GetMousePosition())

Can you check why that code does not work on Mac?

carandraug, 2021-03-07:
> [...] As we no longer have the move-to-mouse option you can only get them back by editing or removing the config file. [...]

I just managed to test on a Mac and it does work. If you select a window in the Windows menu that is outside the current display, it will be moved to the mouse position.  Here's how I tested:

1. start cockpit with two monitors
2. move "Macro Stage" window to the second display
3. close cockpit
4. disconnect second display
5. start cockpit
6. the "Macro Stage" window is not displayed
7. select "Windows>Macro Stage" and the window appears at the current mouse position

One issue I found is that this does not work if the window that's gone is the main window since the main window is not listed on on the "Windows" menu.

iandobbie, 2021-03-09:
You are correct this functionality does seem to work on the Mac I'm not sure why I couldn't get this to work. There is mention of an issue with the "tabbed" mode. I don't have a second monitor at the moment but will try it later today.

carandraug, 2021-03-10:
I have pushed the previously proposed fix. Closing as fixed.

But you're right there's still issues with how saved positions work in tabbed mode. Would also be nice to integrate better the "Windows" menu on the default "Window" menu in Mac (note singular vs plural) and understand what other platforms do when windows get "lost" with multiple monitors. Cockpit can't be the only program with this issue.

closed 2021-03-10 by carandraug

iandobbie, 2021-03-10:
Can we not add the patch that includes the cockpit window in the menu on mac's? I regularly run into this as I swap between my desk with 2 monitors and at home when I generally only have one.