microscope-cockpit/cockpit #717: touchscreen icons don't fit to their buttons

carandraug, 2021-03-03:
The icons in the touchscreen buttons fit nicely inside the buttons on Windows only. In GTK3 and macOS they do not which looks odd.  This how some of them look in GTK3:

[image]

iandobbie, 2021-03-04:
I tried changing the button size to 56x56 rather than 48x48 and it all looked much better on a mac, commit https://github.com/iandobbie/cockpit/commit/e323c7784312bf97de514393fede5f2779c57487

carandraug, 2021-03-08:
Is there no way to avoid hardcoded button sizes?

iandobbie, 2021-03-08:
I think with no size the button should be created exactly the right size to fit its contents.

iandobbie, 2021-03-08:
Yes it does, but not as you might expect. Button are then too wide.
[image]

iandobbie, 2021-03-08:
Can be improved on with a call to 
self.SetWindowStyleFlag(wx.BU_EXACTFIT)
[image]

iandobbie, 2021-03-08:
Still seems to have more room on the right than the left for some reason.