microscope-cockpit/cockpit #718: first image of a mosaic is blank

carandraug, 2021-03-03:
If the first image taken by a camera is part of the mosaic, the image is blank.  Reproduce the issue:

1. start cockpit
2. enable one camera
3. run mosaic on the mosaic window
4. the first image of the mosaic (the centre) will be blank

[image]

If after enabling the camera, one snaps an image before running the mosaic then the issue is gone.

The real issue has to do with the image scaling and not that the first image is not taken. This can be confirmed by clicking "Rescale tiles" on the mosaic window or with the test cameras that are labeled on the corner (the image will appear all white but the number appears.

iandobbie, 2021-03-04:
The image scaling point is obvious in hindsight. The image is grabbed from a  "New Image" event and the scaling is taken from the camera views canvas, but this doesnt calculate a new image scaling until it has received an image. 

Not sure how to deal with it though.

iandobbie, 2021-03-11:
I have a proposed fix for this in commit 87011d8a5b1e1627277651414226b126c6182046

carandraug, 2021-03-12:
> I have a proposed fix for this in commit 87011d8 [...]

The proposed fix checks if scaling is `[0 1]` and if so it assumes that the scaling has not been set (it defaults to `[0 1]`). But the root of the issue is not about the first image, it's about the canvas not having set the scaling. If we ever want the canvas to do something different, such as auto scaling, then the mosaic will always be applying the scaling of the previous image. If being able to display an image the mosaic requires it getting handled in the canvas first, then we should be waiting from an event that the canvas has an image ready and not that the camera sent an image back.

We could have another event from the viewcanvas but I don't think that's the way to go. GUI components should not be dependent on each other, there should be some entity that handles this internal logic. It eventually ends in having an `cockpit.gui.imageViewer.viewCanvas.Image` instance (that's where the `[0.0, 1.0]` default comes) that doesn't actually have an image data.

Anyway, the proposed fix is an hack but I'm not sure how to get it fixed without a lot of changes so let's just make a note of the hack and apply it.

closed 2021-03-13 by carandraug