microscope-cockpit/cockpit #720: Improved documentation.

iandobbie, 2021-03-04:
Martin Hailstone sent me a pull request with substantial improvements to the documentation. Shall I just pull it into master here?

carandraug, 2021-03-04:
Just refer the fork and the branch name on the issue itself (here). I guess we're talking about @hailstonem fork at git@github.com:hailstonem/cockpit.git and its `mh-doc` branch.

carandraug, 2021-03-04:
I looked at the proposed changes to the documentation. Here's some points if others can comment.

1. documentation of userConfig: the changes adds a section about them under a new GUI configuration section. I'm not sure about documenting this userConfig and if we, I don't think we should call it GUI config.

    The reason against calling it GUI config is that userConfig controls things outside the gui part of cockpit such as microscope camera default settings. Also, despite the module being called userConfig is more akin to a cache of the last used settings than actual configuration.

    The reason against documenting it is that things there are more . Comparing with other programs, the stuff saved by userConfig is akin to the default zoom for different sites in Firefox, location of toolbars in office (not sure if office still as toolbars since the new "ribbon" interface), or count of each time a music was played by a music player. Not only are those things that one wouldn't expect to change manually on the file, they are also things that I prefer we keep the freedom to change in the future.

2. README sourcecode file is moved to doc.  This file is very much out of date. It does not even talk about Cocpkit, it's about MUI. We probably should make a lot of changes on it before putting it on the website. We previously just catted the old README to README_sourcecode because there were parts of it that are correct but we didn't had the time to remove the parts that were wrong. And then we made a new README.

3. empty CONTRIB file. If we have no content for it, maybe let's delay its creation until we do. Or since we probably want this on the manual, maybe have it as part of the documentation and not a file on the root of the repository.

4. CC BY-SA license for getting started. We are using GNU FDL on the rest of the documentation, would be nice and cleaner to keep it all under the same license. @hailstonem is that OK? I don't mind changing the previous parts to CC BY-SA .

5. internal links. If a user is navigating the documentation on github, then the links need to be file paths. However, github is the source repository and not the online manual. The `doc/` directory is the source for the online manual (which is created with sphinx) but not the manual itself. The links from one section to another should be using internal hyperlink targets. File paths relative to the repository are meaningless in the manual. Specifically, in `doc/install.rst` we can't use `` `configure your own. <doc/config.rst>` `` to create links to the config section. The README and INSTALL files in the root are not part of the manual so it's fine to use those paths there (but maybe they should be links to the online manual?).

hailstonem, 2021-03-04:
I should preface this with the fact that these are my suggestions for changes, I didn't expect them to be accepted wholesale, as they are obviously not complete, and I did mention to Ian they would need to be checked for correctness.

1. To be honest, I found the current configuration confusing but wasn't really sure how to fix it. Part of that is the naming: there is 'Cockpit Configuration' (an INI file), 'Depot configuration' (a .conf file), and userConfig file: config.py. I might suggest renaming 'Depot configuration' file to .depot, to help with this, and maybe there would be a more specific way to refer to the INI file as well (describing it as general cockpit configuration, when this is clearly only a part of the configuration is probably not sufficient). I take your point about not wanting to document userConfig (which is not called that?), but I think given that the user is directly pointed to the directory where config.py is found (and initially there are no other configuration files to be found), it would be good to at least mention it exists, even if you say it is for internal use only.

2. Currently there is no overview for Cockpit architecture - this was the closest thing I could find. I realise this is a work in progress, but leaving it, without comments, in the main directory as README_sourcecode is confusing, and makes the documentation feel scattered, so I would suggest deleting it completely in the meantime, or at least commenting the fact that it is wildly out of date. The Contrib file was intended to replace the role of this.

3. Up to you.

4. Yes, happy with CC BY-SA or GNU FDL as you prefer.

5. I'm specifically not familiar with how to make RST documentation links, I think I managed to match the internal linking within documents, but wasn't sure about links between documents so they are intentionally links by path. If you want to give an example, I can update the links accordingly - I believe there are only two or three.

hailstonem, 2021-03-04:
Oh, as an aside on point 1, there wasn't an obvious way to change the visible windows in the GUI. Did I miss something? If I accidentally closed windows, then they would not appear on start-up, so the easiest thing was just to edit the userConfig(config.py) file. I am guessing that is not intended behaviour!

carandraug, 2021-03-04:
> I didn't expect them to be accepted wholesale, as they are obviously not complete, [...]

We can wait until they're complete, no problem. Makes it easier to review.

> [...] To be honest, I found the current configuration confusing but wasn't really sure how to fix it. Part of that is the naming: there is 'Cockpit Configuration' (an INI file), 'Depot configuration' (a .conf file), and userConfig file: config.py. 

Both depot and cockpit configuration files are INI files and we expect them to have ".conf" file extension. It's the content of those files that makes them cockpit or depot configuration. The depot file is named such because the internal module that the file configures is also named depot. It's been suggested that we rename it to devices.conf  which will be much more obvious and that the reference to depot is effectively a leak from the internals. What do you guys think?

Martin, that's fresh "untainted" from experience feedback is great, thank you.

> [...] userConfig (which is not called that?) [...]

The file is `config.py` but that's the "configuration" handled by `cockpit.util.userConfig`.

> [...] but I think given that the user is directly pointed to the directory where config.py is found (and initially there are no other configuration files to be found), it would be good to at least mention it exists, even if you say it is for internal use only.

That's a good point, we should say something about it. Also, maybe renaming `config.py` to something else would also help.

> Currently there is no overview for Cockpit architecture - this was the closest thing I could find. I realise this is a work in progress, but leaving it, without comments, in the main directory as [...]

Maybe I should just bite the bullet and update it. Large parts of it I had to write to the paper anyway.

> I'm specifically not familiar with how to make RST documentation links, I think I managed to match the internal linking within documents, but wasn't sure about links between documents so they are [...]

There's very few to fix. I only raised the point because of the issues of navigating the documentation via github (which is not the goal) versus doing it on the final website. It's nice that github shows the rst files properly styled but it's unfortunate that it leads to the expection that then the source repo is the final documentation.

A small aside, the files in the doc directory, they are all part of the same document. The links are for other parts of the same document, even if in source form they are in separate files. We're only exporting the documentation to a muiti page html, but nothing prevents you from exporting the whole thing into a single html page, pdf (it's done via latex), or even TexInfo.

> [...], there wasn't an obvious way to change the visible windows in the GUI. Did I miss something? If I accidentally closed windows, then they would not appear on start-up, so the easiest thing was just to edit the userConfig(config.py) file. I am guessing that is not intended behaviour!

The windows at startup are the windows that were open when you exit cockpit. In the Cockpit "Windows" menu you can select the windows and they will reappear.

carandraug, 2021-03-05:
> Maybe I should just bite the bullet and update it. Large parts of it I had to write to the paper anyway.

I have created a branch to handle only this. It's named "720-doc-sourcecode".

iandobbie, 2021-03-08:
I added some more to my troubleshooting page in my fork of the 720-doc-sourcecode

carandraug, 2021-04-09:
@hailstonem I have merged most of your documentation changes in one form or another. The one where I have the most researvations is your changes to the configuration section. I'm not sure why you restructured it and moved the depot to the top before the cockpit proper config. Is that because the depot configuration is more important? That makes sense but one makes reference to the other.  I also moved your "Configuring Cockpit for the first time" subsection to your new getting started page. I think it fits better there,

I think there's still a lot that could be improved on the documentation. I think we're missing a lot of more practical documentation, what we have is very dry. This should be done by having new documents that read more like guides but I still think it's important to also keep the current reference-type of documentation. See https://documentation.divio.com/

iandobbie, 2023-03-15:
These comments are all 2 years old and  the documentation has been improved enough that I no longer think these comments are relevant. If people have more comments on the documentation we should open a new issue.

closed 2023-03-15 by iandobbie