microscope-cockpit/cockpit #722: licensing - who owns the code really?

carandraug, 2021-03-04:
When we got permission from the department to release the code under GPL, I was under the impression that we could also be the copyright holders of the code.  I have talked with a few people from OUI and they're surprised, the copyright should belong to the university. When I spoke with @iandobbie a couple of weeks ago, he does not remember any agreement about the code belonging to us. @mickp , you were handling this at the time. Could you clarify the situation if you have any documentation? If not, I think this was all misunderstanding and we should transfer copyright to the university as soon as possible (same goes to code in Microscope).

mickp, 2021-03-05:
I'll get back to you shortly.

mickp, 2021-03-05:
I think this is about freedom to sell work outside the GPL: as I understand it, any copyright owner can sell anything they have sole copyright on under different licensing terms. They can not sell code that others have a copyright on without their written agreement.

Take a look at Part B of https://governance.admin.ox.ac.uk/legislation/statute-xvi-property-contracts-and-trusts.
OU asserts a copyright claim to anything produced by staff during their employment, and by students where that work is carried out jointly with staff. If staff members were working for a *department*, different rules may apply. Ultimately, I that comes down to the written agreement between authors and the department, any departmental policies, or any terms in the employment contract.

Either way, neither the university nor the department can assert copyright to any work contributed after the staff members or student left, nor to any contributions from 3rd parties. 

For cockpit, there's the additional complication of the UCSF agreement. I don't know what was finally signed by the lawyers, but what I saw was not a transfer of any rights: it was a non-assertion of rights agreement. I'm not sure OU can assert any claims to anything in the massive initial commit by Chris.

Given that the aim is to keep this open, how about suggesting joint copyright? This might also encourage current  staff to continue contributing if they move elsewhere (e.g. Diamond).

carandraug, 2021-03-05:
> I think this is about freedom to sell work outside the GPL: as I understand it, any copyright owner can sell anything they have sole copyright on under different licensing terms. They can not sell code that others have a copyright on without their written agreement.

We're nowhere near even thinking about selling the code. There is no ongoing talks about that and no one is trying to claim copyright. I've talked with OUI outside the context of work done at Micron. This issue is just about making sure we set the copyright owners correctly while it is, relatively, recent. It was brought up because I thought we had some agreement in place about us being able to claim copyright, been told that it's weird, and now trying to find any documentation on that.

> Take a look at Part B of https://governance.admin.ox.ac.uk/legislation/statute-xvi-property-contracts-and-trusts.
OU asserts a copyright claim to anything produced by staff during their employment, and by students where that work is carried out jointly with staff. If staff members were working for a department, different rules may apply. Ultimately, I that comes down to the written agreement between authors and the department, any departmental policies, or any terms in the employment contract.

Reading that, it seems to me that by default the university claims copyright. A written agreement between the authors and the department is possible but unless that already exists then the university claims copyright. Is there such an agreement in place? I never saw it. Regarding department policies, I'm pretty sure they don't exist. Regarding special terms in the contract, mine says nothing about this, so I guess I fall on the default. What about yours?

> Either way, neither the university nor the department can assert copyright to any work contributed after the staff members or student left, nor to any contributions from 3rd parties.

Yes, of course. But at the moment, all the code in these projects was written while we were employed by the university. So unless we have a document saying otherwise, I think it belongs to the university.

> For cockpit, there's the additional complication of the UCSF agreement. I don't know what was finally signed by the lawyers, but what I saw was not a transfer of any rights: it was a non-assertion [...]

That code is owned by UCSF and under 3 clause BSD which is quite permissive.

> Given that the aim is to keep this open, how about suggesting joint copyright? This might also encourage current staff to continue contributing if they move elsewhere (e.g. Diamond).

Yes, that would be nice.

mickp, 2021-03-05:
I didn't expect they're looking at selling any part of the code now: my point was that OUI can not (at any point in the future) sell any part of the code over which some other entity asserts copyright under different licensing terms.

I think the OU-UCSF agreement just states that USCF will not assert their rights. I do not know if that means that OU can assert any rights over work done at UCSF. There's contributions from Julio and Tom where ownership is unclear.

We started adding copyright notices where we added or modified code. I don't recall anything specific in my contract about copyright, which I believe means the university can assert copyright over my contributions. However, Diamond may also have a claim as much of the work was undertaken using their equipment.

mickp, 2021-03-05:
Right. Found my contract with Diamond. They just ask for a free, unrestricted, perpetual license, so I don't believe they can assert a copyright claim.

carandraug, 2021-03-06:
> [...] I don't recall anything specific in my contract about copyright, which I believe means the university can assert copyright over my contributions. [...]

I believe this is the root of the issue. This won't be a case of the university later asserting their copyright. For them to do it later it means they are the copyright holder now and that it is us who are wrong by asserting copyright now. It would be nice to get that clarified now.

> There's contributions from Julio and Tom where ownership is unclear.

@juliomateoslangerak and @tomparks : do you know who is the copyright holder of the code you wrote for Cockpit?

iandobbie, 2021-03-16:
Seems pretty clear the university owns code written by the staff here. However, I think there are enough other contributors that trying to relicense the code would be basically impossible. I think we should just try fix the copyright headers, but maybe move the current copyright claims into an author line for OU staff so people get the credit they deserve.

carandraug, 2021-04-09:
@juliomateoslangerak in #677 you wrote:
> In principle, the owner of anything I write during my working ours is the CNRS. But I ignore all the details.

Which I think was an accident and you meant to the discussion in this issue.

juliomateoslangerak, 2021-04-09:
Indeed that was meant to be here. 
Do you need me to get details?
We also signed an MTA back in the day.

carandraug, 2021-04-09:
> Seems pretty clear the university owns code written by the staff here.

Yes. I just spoke with OUI about this and asked whether the university would accept to assign the copyright to the individual people. Basically, since this is an open source project and there's no money to be involved, the university is likely to agree. However, this needs to be done in a per individual case, i.e., if anyone wants to have the copyright of their contributions assigned to them, then they will need to get their own agreement with the department. We can't get a blank letter about all contributions to the project (but Ilan could get a letter from the department that covers everyone on his group).

So, unless anyone is planning on requesting that for themselves (I'm not going to ask this for me), here's what I propose:

1. all copyright currently assigned to Ian Mick, and me I will assign to the University of Oxford instead
2. @dstoychev contributions so far are in the touchscreen and he's done that without supervision from staff. So I'm leaving copyright to him.
3. @NickHallPhysics merged contributions so far are on the stageMover and on the SIMFlux experiment, which was with supervision so I'll assign it to both Oxford and him. I'll guess it's the same for @langefj (fixed issues in the mosaic window).
4. I'm not sure what to do about @tomparks contributions. i guess that's under supervision of University staff (Mick) so I will assign to him and to Oxford (unless his Diamond contract said otherwise).
5. @juliomateoslangerak should I use "Centre national de la recherche scientifique (CNRS)" as the copyright holder of your work?
6. Add an authors section below the copyright notice with the name of the people that actually wrote the code.

mickp, 2021-04-09:
Sounds good to me. I mentioned the agreement with Diamond above - they assert no claim, so Tom -> Oxford is fine, I think.

Maybe add a `CONTRIBUTORS` file?

dstoychev, 2021-04-09:
Thanks for taking care of this! I do not need the copyright, feel free to assign it to the University of Oxford.

carandraug, 2021-08-20:
Here's the proposed authors file. If no one has any objections, I will push this next week:

```
Many people have contributed to the development of Cockpit.  The
following list, sorted alphabetically, includes their contacts as well
as a short description of their contributions:

- Ana Rita Carvalho Faria <anarita.cfaria@gmail.com>

    Bug fixes.

- Christopher Weisiger <christopher.weisiger@gmail.com>

    Wrote the first version of Cockpit.

- Danail Stoychev <danail.stoychev@exeter.ox.ac.uk>

    Wrote the new touchscreen interface as well as multiple bug fixes.

- David Miguel Susano Pinto <david.pinto@bioch.ox.ac.uk>

    Been involved in different parts of Cockpit since 2017.

- Eric Branlund <eric@msg.ucsf.edu>

    Fixed multiple bugs in Cockpit while at UCSF and handled the
    handover from UCSF to Micron Oxford.

- Frederik Lange <frederik.lange@dtc.ox.ac.uk>

    Bug fixes.

- Ian Dobbie <ian.dobbie@bioch.ox.ac.uk>

    Been involved in different parts of Cockpit since 2015.

- Julio Mateos-Langerak <julio.mateos-langerak@igh.cnrs.fr>

    Been involved in different parts of Cockpit since 2015.

- Martin Hailstone <martin.hailstone@engs.ox.ac.uk>

    Multiple improvements to the documentation.

- Mick Phillips <mick.phillips@gmail.com>

    Been involved in its development since 2015, having effectively
    led its development until 2020.

- Nicholas Hall <nicholas.hall@dtc.ox.ac.uk>

    Wrote the SIMFlux experiment code.

- Sebastian Haase <sebastian.haase@fu-berlin.de>

    Wrote the original Cockpit code as well as multiple util modules
    which while part of Priithon have also been vendorised in Cockpit.

- Thomas Park <thomasparks@outlook.com>

    Wrote the first test suite for Cockpit.
```

carandraug, 2021-08-20:
See my 722-fix-licensing branch.

For future reference

    find cockpit -name '*py' | xargs sed -ie 's/\#\# Copyright (C) .*/## Copyright (C) 2021 University of Oxford/' cockpit/__init__.py
    find cockpit -name '*.py' | xargs sed -i '$!N; /^\(\#\# Copyright (C) .*\)\n\1$/!P; D'

Followed by manually changing those few that should not be assigned to the University of Oxford.