microscope-cockpit/cockpit #723: lights panel with colour does not work on Mac

ARCFaria, 2021-03-07:
The thin line under the light name that is supposed to show the light wavelength is always grey on a Mac.

[image]

iandobbie, 2021-03-08:
The camera colour indicators also dont work. The title bar in the camera views works but the bar under the camera name in the main window does not take a colour.

carandraug, 2021-03-08:
This is currently done with:

        line = wx.StaticBox(self, size=(-1,4), style=wx.LI_HORIZONTAL)
        line.SetBackgroundColour(wavelengthToColor(self.light.wavelength))

Which creates an empty control with a dark grey line around (the static box) and colours the background. If one looks closely on the screenshoot above, one can see the dark grey colour from the grey colour as well as a lighter grey which is the outside of the static box. Because the grey line is not the background, it is not affected by the `SetBackgroundColour`. This works in Linux and Windows because there's a margin between the start of control and the actual static box line, which is larger than the size we're giving to the whole control. See in Linux if we increase its height to 20:

[image]

There might still be a problem with mac ignoring `SetBackgroundColour` but that's not the whole thing.

At least for starter, replacing the use of `wx.StaticBox` with simply `wx.Control` might help. If that does not work in Mac, then Mac really is ignoring `SetBackgroundColour`. I don't think there's any wx control for this only, so maybe we would need a `wx.ClientDC` that we paint whatever colour we want.

@ARCFaria can you try my "723-control-for-colour" branch and see if that fixes it?

iandobbie, 2021-03-08:
[image]
Seems to work on my mac

iandobbie, 2021-03-08:
Obviously also needs to be applied to the cameras as well, although these might not have colours.

carandraug, 2021-03-08:
Thank you for testing. I have pushed the fix to master. It should also fix the colour under the cameras (but that needs a drawer or filterwheel device to set a colour for the camera. Such dummy devices are not part of the dummy devices created automatically by cockpit which is why they are appearing grey on your case).

Closing as fixed.

Since I was looking at the code I also changed the height of the line to be function of the font size (instead of hardcoded to 4 pixels).

closed 2021-03-08 by carandraug