microscope-cockpit/cockpit #725: Display last Experiment in debug only runs once.

iandobbie, 2021-03-09:
How to reproduce:

1) create and run an experiment then display the action table plot from the debug window.
2) close the action table plot
3) create a new experiment and try to plot it. 

If you don't close the initial plot then a new plot will be created but if you close the forst window trying to open the second crashes things: 

Traceback (most recent call last):
  File “/Users/ian/src/cockpit/cockpit/handlers/executor.py”, line 592, in _OnDisplayLastExperiment
    plot_action_table_profile(experiment.lastExperiment.table)
  File “/Users/ian/src/cockpit/cockpit/handlers/executor.py”, line 530, in plot_action_table_profile
    plt.show(block=False)
  File “/Users/ian/Library/Python/3.9/lib/python/site-packages/matplotlib/pyplot.py”, line 353, in show
    return _backend_mod.show(*args, **kwargs)
  File “/Users/ian/Library/Python/3.9/lib/python/site-packages/matplotlib/backend_bases.py”, line 3524, in show
    manager.show()  # Emits a warning for non-interactive backend.
  File “/Users/ian/Library/Python/3.9/lib/python/site-packages/matplotlib/backends/backend_wx.py”, line 1049, in show
    self.frame.Show()
RuntimeError: wrapped C/C++ object of type FigureFrameWxAgg has been deleted

carandraug, 2021-03-10:
I can't reproduce this issue on Linux. One thing I noticed is that my backend is wx_agg and not wx. See also notes in #650 about letting pyplot pick the be backend automatically.

iandobbie, 2021-03-12:
I tried to add 
 import matplotlib
 matplotlib.use('WXAgg')
 import matplotlib.backends.backend_wxagg
to the imports of handlers/executor.py but still got 

Traceback (most recent call last):
  File “/Users/ian/src/cockpit/cockpit/handlers/executor.py”, line 595, in _OnDisplayLastExperiment
    plot_action_table_profile(experiment.lastExperiment.table)
  File “/Users/ian/src/cockpit/cockpit/handlers/executor.py”, line 533, in plot_action_table_profile
    plt.show(block=False)
  File “/Users/ian/Library/Python/3.9/lib/python/site-packages/matplotlib/pyplot.py”, line 353, in show
    return _backend_mod.show(*args, **kwargs)
  File “/Users/ian/Library/Python/3.9/lib/python/site-packages/matplotlib/backend_bases.py”, line 3524, in show
    manager.show()  # Emits a warning for non-interactive backend.
  File “/Users/ian/Library/Python/3.9/lib/python/site-packages/matplotlib/backends/backend_wx.py”, line 1049, in show
    self.frame.Show()
RuntimeError: wrapped C/C++ object of type FigureFrameWxAgg has been deleted

Not sure about the error as it seems to have a WxAgg object but the python error above is from backends/backend_wx.py

carandraug, 2021-04-01:
Which version of matplotlib are you using? I've tried matplotlib 3.4.1 and 3.2.2 (but I'm also on Linux).

iandobbie, 2021-04-01:
I seem to have 3.3.4, currently on python 3.9.1

carandraug, 2021-04-01:
I can reproduce the issue.  I'm thinking this is a bug in the 3.3.x series of matplotlib (works in 3.2.2, fails in 3.3.0 and 3.3.4, then works again in 3.4.0).

carandraug, 2021-04-01:
I have found the issue in matplotlib changelog. This is issue matplotlib/matplotlib#17769 which was fixed in matplotlib/matplotlib#19596 . I guess there's always thing we can do to make it work on ours side, worst case scenario we can monkey patch matplotlib. However, this is a minor feature and I think it's just simpler to let it be and close this as "invalid".

iandobbie, 2021-04-01:
Ok I’ll upgrade to 3.4 and check it works then close issue as matplotlib bug.

iandobbie, 2021-04-01:
Obviously a matplotlib bug. Upgrading to 3.4.1 fixed this on my mac.

closed 2021-04-01 by iandobbie