microscope-cockpit/cockpit #726: "cannot schedule new futures after shutdown" error at exit

carandraug, 2021-03-09:
In Linux I sometimes get the following error at exit:

```
Exception in thread _updater:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/carandraug/src/cockpit/cockpit/handlers/lightPower.py", line 92, in _updater
    queries[light] = executor.submit(getPower)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 151, in submit
    raise RuntimeError('cannot schedule new futures after shutdown')
RuntimeError: cannot schedule new futures after shutdown
```

This is very much non reproducible, it happens rarely. It just happened when starting cockpit with `cockpit --no-config-files` so this is not because of issues on a device server.

iandobbie, 2023-02-07:
I have also found this happening on my mac. I guess we need some structure around the getpower calls to ensure that they don't schedule futures after a shutdown. Maybe the shutdown process is trying to update the data and calling the future in the shutdown process.

iandobbie, 2023-07-07:
Closing as duplicate of #865

closed 2023-07-07 by iandobbie