microscope-cockpit/cockpit #727: Cannot resize exception box to read errror message

iandobbie, 2021-03-10:


carandraug, 2021-03-10:
This issue is Mac specific. The window is as large as the message bug the section with the traceback is too small and the window can't be resized.

iandobbie, 2021-03-15:
I have added the wx.RESIZE_BORDER to the dialog style which at least allows resizing to see the message. branch 727-mac-resize-exception

iandobbie, 2021-03-31:
Finally fixed properly. As in  windows the mac text sizer ignore newlines. Fix was to just add WXMAC to the relevant if statement so that the mac also explicitly counts the lines

closed 2021-03-31 by iandobbie

carandraug, 2021-03-31:
Adding the `wx.RESIZE_BORDER` does make it possible to resize the frame but it removes the rest of the default style flags:

```
>>> format(wx.DEFAULT_DIALOG_STYLE, 'b')
'100000000000000001100000000000'
>>> format(wx.RESIZE_BORDER, 'b')
'1000000'
```

I fixed this by OR the two flags.

iandobbie, 2021-03-31:
Oh right. As no other flags were explicitly set I assumed it was ok. Thanks for fixing it.

I actually pulled a further fix which was much better today. But I decided to leave the resize border in case of issues where it is needed

Ian

>>> format(wx.DEFAULT_DIALOG_STYLE, 'b')
'100000000000000001100000000000'
>>> format(wx.RESIZE_BORDER, 'b')
'1000000'

I fixed this by OR the two flags.