microscope-cockpit/cockpit #729: goto XYZ box needs a cancel button.

iandobbie, 2021-03-10:
If you right click in the macrostage window and get a goto XYZ box it has no cancel button. If you accidentally erase the existing values you might have no easy way to avoid a unknown movement in some direction which could have serious consequences.

iandobbie, 2021-03-11:
Fixed in my branch 729-goto-cancel the code was already there just commented out.

carandraug, 2021-03-11:
I think you forgot to push that branch.

You're right the code that adds a cancel button is there, but the reason it is commented is that it was not being checked. JUst uncommenting it will add the cancel button back but the rest of the code will not check for it so it will be same as pressing "Okay". The commenting happened in 8b6a3c52af9a31d216bf .

iandobbie, 2021-03-11:
No David. I did actually test it. If you change the values and push cancel the stage does not move but if you press ok it does. I think there must be some wx magic in there as the code doesn’t really do anything just add the cancel button

Sorry if I did t push it. I’ll do it tomorrow and do some more tests. Maybe with a real system rather than just the simulation.

Ian

carandraug, 2021-03-11:
> No David. I did actually test it. If you change the values and push cancel the stage does not move but if you press ok it does. [...]

Or maybe it's a Mac thing? I tested uncommenting the three lines that added the cancel button and added a print statement to the values returned by `getManyNumbersFromUser`:

```
@@ -408,6 +446,7 @@ class MacroStageXY(macroStageBase.MacroStageBase):
                 "Go To XYZ",('X','Y','Z'),
                 position,
                 atMouse=True)
+        print(values)
         newPos=[float(values[0]),float(values[1]),float(values[2])]
 #Work out if we will be ouside the limits of the current stage
         posDelta = [newPos[0]-position[0],newPos[1]-position[1],newPos[2]-position[2]]
```

I then do:

1. start cockpit and right click on the macro stage canvas, the dialog opens
2. enter a value of zero on all three text boxes, and press three
3. the print statement shows that the function returned `[0, 0, 0]`
4. the stage position coordinates, as shown in the text boxes of the macro stage, also show position 0 for all axis

carandraug, 2021-03-12:
I just wrote a new dialog specific for this purpose. Gives us flexibility to add extra things such as limiting the value to the stage limits. Could probably be improved. Check my branch "729-goto-xyz-new-dialog".

iandobbie, 2021-03-12:
Looks good and works fine on the mac.

carandraug, 2021-03-12:
I noticed that on the touchscreen there was an exact copy of the code (right click on the macro stage but there is also a button of its own on the `StageControlCommon` panel). So I first made a function with the duplicated code and then added the new dialog to a single place. Closing as fixed.

closed 2021-03-12 by carandraug