microscope-cockpit/cockpit #730: windows outside the display at startup

carandraug, 2021-03-10:
When cockpit starts it places the window at the same location as they were when cockpit last exited. This is a problem if those window coordinates are no longer valid. To reproduce:

1. start cockpit with two displays.
2. move all windows to the secondary display and then exit cockpit
3. start cockpit and the windows will be placed outside the main display

What happens then is system dependent. On Mac, the windows will be inaccessible. I'm not sure what happens on Windows. On Linux, the windows will be placed in the closest valid position.

Individual windows can be moved back by selecting them from the "Windows" menu in the menu bar. However, on Windows this requires access to the main window which may be the missing window. In Mac, the menu bar is at the top of the display but the main window is not listed there.

A suggestion is to list the main window on the "Windows" menu but that fixes for mac only. It also only fixes the main window and one still needs to select the other windows one by one.

There is no problem if the secondary display is removed while cockpit is running as in all cases the windows are all automatically moved to a valid position. The problem is only on `CockpitApp.SetWindowPosition` which uses invalid positions. I think we can check for validity of those coordinates and if invalid place the window at the centre of the main display.

carandraug, 2021-03-10:
Proposed fix on my "730-centre-window" branch. To be tested on mac.

iandobbie, 2021-03-10:
Seems to work on my system. When started from a position on an external monitor all windows start centered on the main screen. It is a bit of  strange position but definitely solves the problem.

carandraug, 2021-03-10:
> It is a bit of strange position but definitely solves the problem.

I agree that it doesn't fell a natural location but I don't know what would be. The logic of where to place new windows (outside Cockpit) seems to be dependent on the window manager and that at least on my system it also takes into account the location of any other window already open. I think with an empty screen, the first window appears somewhere on the top left quadrant but not on the corner.

Where would you have them placed? We can position them wherever we want but we need to compute that position. We can have wx pick the "natural" position only when creating the window. Afterwards, we need to specify it.

carandraug, 2021-03-11:
> [...] We can have wx pick the "natural" position only when creating the window. Afterwards, we need to specify it.

It occurred to me that we can just skip the whole `SetPosition` if the position is invalid to rely on the "native" positioning of the window. I pushed another commit to the "730-centre-window" branch. Does that feels better? Note that this also affects the behaviour of "Reset window positions" menu.

iandobbie, 2021-03-11:
I think this is slightly less strange, but to be honest the earlier solution solved this rather rare edge case. It is just one that strikes me frequently. I often move between 1 and 2 monitors and am frequently stopping and restarting cockpit. I would close this issue as is.

carandraug, 2021-03-13:
Merged the fix that leaves the windows on the "native" place. Closing as fixed.

closed 2021-03-13 by carandraug