microscope-cockpit/cockpit #733: "snapped" images should appear on the mosaic view

carandraug, 2021-03-12:
The point of the mosaic is to create a map of the stage. However, the only images that appear on the mosaic canvas are images acquired via the "run mosaic" option. Images acquired by pressing the "Snap" button on the main window (will that be the most common situation?) do not appear but they can be just as useful to map the stage.

iandobbie, 2021-03-12:
Yes these images can be useful and there is a keyboard shortcut to transfer them but many images just snapped to find focus or set exposure times/laser power will just corrupt the mosaic so it is a specific design choice not to add the images to the mosaic by default.

iandobbie, 2021-03-15:
Maybe we need a more transparent method to get the images into the mosaic but I think autoloading all of them is very bad idea from a usability point of view.

carandraug, 2021-03-30:
> Yes these images can be useful and there is a keyboard shortcut to transfer them [...]

I did not know about this.  I looked into the source code and found that this is the decimal character in the numpad (`.`). This works from the main window and the msoaic window but not the from the camera view window.

iandobbie, 2021-03-31:
Good point. There are two issues I see here:
1) Another control that has very limited discoverability. I only found it by reading the code and now use it pretty regularly. 
2) We should add the functionality to the camera views, maybe both the keyboard shortcut and maybe add it to the right click menu.

iandobbie, 2022-03-11:
I am trying to fix up a few issues and this seems like a no brainer. Adding it to the right click menu is very easy and I will do this.

iandobbie, 2023-03-15:
Added to camera view context menu in 7045e18e283a7dfc4ef11e63379fe98856d18618

closed 2023-03-15 by iandobbie