microscope-cockpit/cockpit #734: Experiments only save the last image

dstoychev, 2021-03-16:
Whenever I try to run an experiment, the only data that's saved is the very last image of the whole sequence. I tried this with both the regular Z-Stack and the stuttered Z-Stack experiments, and the result is the same. While debugging, I moved the starting of the run thread after the save thread is already started, and this seems to have partially fixed the problem. However, now I get a timeout error and about half as many slices as I would expect, although visually the stack seems alright, so I suspect either skipped images or larger slice height than specified.

iandobbie, 2021-03-16:
There is obviously a major problem here. We have several system that are running experiment successfully so I am very surprised that you are not getting anything. We have seen timeout regularly on the red pitaya but the experiments seem to run fine. The stacks half the expected size is almost certainly attempting to trigger the
Camera before it is ready to receive a trigger. Many cameras have an output signal to indicate when they are ready to trigger. You just need to monitor the camera trigger and this output on an oscilloscope.

Ian

iandobbie, 2021-03-17:
Another useful feature recently added (you may have to update to get it) is the dsp debug window has a button to show  the last experiment action table in graphical form ("Display last experiment"). This should basically match your oscilloscope traces.

dstoychev, 2021-03-17:
The code has been working so far because of convenience, but there is still a timing problem which is easily triggered if you deviate from the norm even slightly.

In order for the `DataSaver` to get the experiment images, it needs to have its `startCollecting()` method called, so that it subscribes to the `NEW_IMAGE` event. The problem is that the thread which runs the experiment is started before this method is called. This means that all data, received in the interval between these two events, is lost. Typically, this interval will be negligeably small, e.g. less than 10ms, so there is no data loss. However, even a small delay is enough to break the delicate balance. Logically, the running thread should be called after the `startCollecting()` method.

carandraug, 2021-03-30:
You're right @dstoychev . Looking in the history, I find this is a regression introduced in 7f8ab9ffd4d9d6a1899c 

I have now fixed this issue by moving the start on the run thread to after the saver thread is configured and started (just like it was before). Closing as fixed.

closed 2021-03-30 by carandraug