microscope-cockpit/cockpit #735: Trigger type of light sources

dstoychev, 2021-03-17:
python-microscope/microscope@ed883b5270c58dca9af0ed11a7ac190928658900 introduced trigger types for light source devices. At the moment, Cockpit doesn't have the means to change this. Not all `LightHandler` objects need it. For example, `AmbientLight` doesn't.

iandobbie, 2021-03-17:
Davids argument was that no light sources needed this. If you provide a hardware trigger then the light always uses this from cockpit and you setup microscope to configure the device as HW triggered. mostly ok, but things like the AO device wanting to SW trigger items mean this isn't necessarily always the best approach.

dstoychev, 2021-03-18:
I don't know how AO devices work, so I cannot comment on that topic, but I wanted to say that this issue should also allow the user to specify a default trigger type in the configuration.

dstoychev, 2021-08-14:
This is now implemented in https://github.com/dstoychev/cockpit/tree/lights_settings.