microscope-cockpit/cockpit #739: Error drawing view canvas on Linux with wxPython 4.1.1

carandraug, 2021-04-06:
On Linux, with wxPython 4.1.1, the view canvas always fails to display and the following appears on stdout:

```
Error drawing view canvas: Attempt to retrieve context when no valid context
```

while on stderr shows the traceback:

```
  File "/home/carandraug/.local/bin/cockpit", line 11, in <module>
    load_entry_point('microscope-cockpit', 'gui_scripts', 'cockpit')()
  File "/home/carandraug/src/cockpit/cockpit/__init__.py", line 389, in _setuptools_entry_point
    return main(sys.argv)
  File "/home/carandraug/src/cockpit/cockpit/__init__.py", line 349, in main
    app.MainLoop()
  File "/home/carandraug/.local/lib/python3.7/site-packages/wx/core.py", line 2237, in MainLoop
    rv = wx.PyApp.MainLoop(self)
  File "/home/carandraug/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py", line 660, in onPaint
    traceback.print_stack()
```

Printing the trace from where the exception is caught is not very helpful, so I changed it so it printed the trace of the exception itself. Which now shows:

```
Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py", line 629, in onPaint
    self.image.draw(pan=(self.panX, self.panY), zoom=self.zoom)
  File "/home/carandraug/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py", line 323, in draw
    (-hlim + i*dx, -vlim + j*dy)] )
  File "src/latebind.pyx", line 39, in OpenGL_accelerate.latebind.LateBind.__call__
  File "src/wrapper.pyx", line 321, in OpenGL_accelerate.wrapper.Wrapper.__call__
  File "/home/carandraug/.local/lib/python3.7/site-packages/OpenGL/arrays/arrayhelpers.py", line 156, in __call__
    contextdata.setValue( self.constant, pyArgs[self.pointerIndex] )
  File "/home/carandraug/.local/lib/python3.7/site-packages/OpenGL/contextdata.py", line 58, in setValue
    context = getContext( context )
  File "/home/carandraug/.local/lib/python3.7/site-packages/OpenGL/contextdata.py", line 41, in getContext
    """Attempt to retrieve context when no valid context"""
OpenGL.error.Error: Attempt to retrieve context when no valid context
```

juliomateoslangerak, 2022-01-12:
Until this is fixed, should we pin down the version of wxPython to 4.1.0. Currently it is =<4.1

carandraug, 2024-06-08:
If we pin to wxPython<=4.1 then we can't install with recent versions of Python.

I've dug a bit on this now and it seems that part of the issue is ours. On Linux (with GTK), we set the `GDK_BACKEND` to `x11`  because wxglcanvas does not work with wayland (see #347 ). But that issue was fixed in wxWidgets 3.1.5 (I guess that's what's included in wxPython 4.1.1 which tracks the wxWidgets master branch).

The fix in wxWidgets was done by adding support for EGL. But this works by changing how wxWidgets is built --- either EGL for Wayland or GLX for X11. If you use EGL/X11 or GLX/x11 then things do not work. The default builds now use EGL but we are still forcing X11. In theory, that is the first thing we are doing wrong.

Another issue is PyOpenGL which may be runing with `egl` or `x11` (should be possible to set it by setting `PYOPENGL_PLATFORM=egl`). It seems that to make things work, the build of wxWidgets, the GDK backend, and the OpenGL options all need to match but it's not clear to me how.

When trying to fix this issue, remember we are forcing this:

```
    if wx.Platform == '__WXGTK__' and 'GDK_BACKEND' not in os.environ:
        os.environ['GDK_BACKEND'] = 'x11'
```

See related issues:

- wxWidgets/wxWidgets#17702 - in wxWidgets where the wxglcanvas issue and fix is discussed
- wxWidgets/Phoenix#1923 - in wxPython where clarification of state is requested and where I ask if it is possible to check at runtime if wxWidgets was built with EGL or GLX

carandraug, 2024-06-08:
I was previously using wxPython 4.2.0 and PyOpenL from Debian repositories.  I didn't confirm it but found online some people saying that they were built for GLX (x11 only).

I have now installed wxPython 4.2.2 from PyPI and it's all working now. The `GDK_BACKEND=x11` described in #347 is also no longer needed so I'm removing it. It would probably be required for older versions of wxPython but I'm not sure where to draw the line. I don't want to bump the requirement for newer versions of wxPython though, people can set those env variables themselves. But if someone feels that we should, please go ahead.

I'm closing this issue as something that upstream has now fixed (but a user still needs to be careful to mix and match the right builds of the dependencies)

closed 2024-06-08 by carandraug