microscope-cockpit/cockpit #745: Buttons clicked appear as a list inside Windows menu

ARCFaria, 2021-05-04:
I mentioned this to @carandraug last week. I understood you're aware of this, but as I couldn't find an issue about it, I'm just pasting here the screenshots of what I'm referring to - see below.

this is the Windows menu, when you open cockpit:

[image]

And then this is the list after clicking camera settings, single and multi-site experiment:
[image]

After some time using the microscope and repeating the same type of experiments, these items just accumulate and we end up with a big list of repeated new "windows".

iandobbie, 2023-09-27:
This is still and issue, we should work out how to remove these redundant windows from the menu, especially multiple copies of the same window.

iandobbie, 2024-10-03:
So it looks like the experiment menu items get added a single time and reused, however at least some other menus, eg generate sim sequence and set Sim diffraction angle, generate a new window entry for every time they are used.

carandraug, 2024-10-09:
I've looked at this again. I think what we want in the menu bar are entries to:

- show "core" cockpit windows
- show extra cockpit windows such as csv plotter, debug windows, SIM intensity profile

Things we don't want to have listed are:

- ancillary hidden windows such as the one used by AuiManager for hint on the logging window (see #617)
- dialog windows (whether they really are `wx.Dialog` such as those in `getNumberDialog` or are `wx.Frame` doing the job of a dialog such as device settings windows)

This issue is specifically about those dialog windows. At the root of it all is that we are unable to automatically distinguish between those 4 class of windows.

So maybe we could just have a `LIST_AS_COCKPIT_WINDOW` attribute on windows that we want shown and ignore others. Then the menu bar could keep itself up to date correctly (for the extra cockpit windows such as csv plotter, debug windows, SIM intensity profile, we could continue to have semi-hardcoded entries like what we already do now).

Ideally though, we would have no state on the windows (e.g., last image acquired is stored in the `CamerasWindow` instead of the imager and log messages are stored on the `LoggingWindow`) so windows could be closed properly instead of hidden, and we would list what windows shuld be available to cockpit configuration.

--- 
I've also looked at having the cockpit app keeping a list of windows and have windows register with them when they are created but then we would need to have the windows created first and we really want to go in the direction of not having to create all windows.

carandraug, 2024-10-14:
This is now fixed on 9ccdb70ee17c97d71655934efd2c6887075bbe84 Closing as fixed.

closed 2024-10-14 by carandraug