microscope-cockpit/cockpit #750: Plot last action table fails on SIM experiments.

iandobbie, 2021-05-21:
Not investigated but at diamond the plot last action table works fine for Z stacks but fails on a SIM
stack. Here is the error output

Traceback (most recent call last):
  File "c:\b24\cockpit\cockpit\handlers\executor.py", line 590, in _OnDisplayLastExperiment
    plot_action_table_profile(experiment.lastExperiment.table)
  File "c:\b24\cockpit\cockpit\handlers\executor.py", line 522, in plot_action_table_profile
    axs[i].step(data[0], data[1], where="post")
  File "C:\python37\python-3.7.2.amd64\lib\site-packages\matplotlib\__init__.py", line 1810, in inner
    return func(ax, *args, **kwargs)
  File "C:\python37\python-3.7.2.amd64\lib\site-packages\matplotlib\axes\_axes.py", line 2064, in step
    return self.plot(x, y, *args, **kwargs)
  File "C:\python37\python-3.7.2.amd64\lib\site-packages\matplotlib\__init__.py", line 1810, in inner
    return func(ax, *args, **kwargs)
  File "C:\python37\python-3.7.2.amd64\lib\site-packages\matplotlib\axes\_axes.py", line 1611, in plot
    for line in self._get_lines(*args, **kwargs):
  File "C:\python37\python-3.7.2.amd64\lib\site-packages\matplotlib\axes\_base.py", line 393, in _grab_next_args
    yield from self._plot_args(this, kwargs)
  File "C:\python37\python-3.7.2.amd64\lib\site-packages\matplotlib\axes\_base.py", line 370, in _plot_args
    x, y = self._xy_from_xy(x, y)
  File "C:\python37\python-3.7.2.amd64\lib\site-packages\matplotlib\axes\_base.py", line 205, in _xy_from_xy
    by = self.axes.yaxis.update_units(y)
  File "C:\python37\python-3.7.2.amd64\lib\site-packages\matplotlib\axis.py", line 1473, in update_units
    default = self.converter.default_units(data, self)
  File "C:\python37\python-3.7.2.amd64\lib\site-packages\matplotlib\category.py", line 103, in default_units
    axis.set_units(UnitData(data))
  File "C:\python37\python-3.7.2.amd64\lib\site-packages\matplotlib\category.py", line 169, in __init__
    self.update(data)
  File "C:\python37\python-3.7.2.amd64\lib\site-packages\matplotlib\category.py", line 186, in update
    for val in OrderedDict.fromkeys(data):
TypeError: unhashable type: 'numpy.ndarray'

ARCFaria, 2021-08-16:
The same happened on DeepSIM for SIM experiments, but I didn't check for any other experiment type. 
Any hint on what is the problem behind this?

juliomateoslangerak, 2021-08-17:
I will try to reproduce this and have a look

carandraug, 2021-08-17:
All DeepSIM packages were updated recently and I assume the same happened in CryoSIM. If this was not an issue previously, my first guess would be that something changed in recent version of matplotlib.

juliomateoslangerak, 2021-08-17:
I reproduced the issue. The problem is that the plots actions table cannot deal with actions that are neither digital or analogue. That is the case of, at least the polarization rotator. I have a fix here where the plotting is just ommiting actions if they cannot be easily classified as analogues or digitals. Might be interesting to look one day at ways to represent other types of actions in the same plot. I guess it would be up to the device to define a representation in a plot...

the fix is here https://github.com/juliomateoslangerak/cockpit/commit/2c735b2430ce7f0dad11319c7da7dc244d0f6f44

iandobbie, 2021-08-17:
Good catch Julio, but I am surprised that the pol rotator events are not just analog events as that is how the pol rotator is controlled on both deepsim and cryosim

juliomateoslangerak, 2021-08-17:
Indeed. They must be converted to an analogue downstream, but at this point in time the actions are angle and wavelength tuples `(0, 488.0)`.
Would it make sense to define handler specific plotting functions? plot_action_table_profile could try to use this function and fall back to common digital or analogue if not implemented...
In general, it would be nice to see the polarization rotator in the plot

iandobbie, 2021-08-17:
Seems to me that the action table must not quite be complete as the pol rotator must replace the triple before it is uploaded to the dsp (or equiv). I’ll see if I can understand what the issue is.

Ian