microscope-cockpit/cockpit #751: intensity profile plot problem.

iandobbie, 2021-05-21:
The intensity profile tool isnt working.

Traceback (most recent call last):
  File "c:\b24\cockpit\cockpit\util\intensity.py", line 416, in loadFile
    xpos, ypos = self.profiler.guessBeadCentre()
  File "c:\b24\cockpit\cockpit\util\intensity.py", line 165, in guessBeadCentre
    3*nx / 8 : 5*nx / 8]
TypeError: slice indices must be integers or None or have an __index__ method
Traceback (most recent call last):
  File "c:\b24\cockpit\cockpit\util\intensity.py", line 416, in loadFile
    xpos, ypos = self.profiler.guessBeadCentre()
  File "c:\b24\cockpit\cockpit\util\intensity.py", line 165, in guessBeadCentre
    3*nx / 8 : 5*nx / 8]
TypeError: slice indices must be integers or None or have an __index__ method

juliomateoslangerak, 2021-05-21:
This is very similar to #558. Incompatibility with python 3 I did some modifications and since then it was working for me.

carandraug, 2021-05-22:
> This is very similar to #558.

More than very similar, this issue seems to have been fixed as part of that. The line with the issue `3*nx / 8 : 5*nx / 8` has since been changed to return integer.

@iandobbie maybe close as "already fixed"?

iandobbie, 2021-05-23:
Already fixed. Diamond cryosim is just behind the times.

closed 2021-05-23 by iandobbie