microscope-cockpit/cockpit #752: It is possible to open two copies of cockpit simultaneously.

iandobbie, 2021-05-28:
This used to be prevented. We managed to have some very weird behaviour where a z stack would work but a sim stack wouldn't. Turned out the second copy of cockpit running was killing the SIM stack, not exactly sure why, but would should at least warn on startup of a second copy. Also not sure how to do this, maybe check on server address binding?

Ian

carandraug, 2021-08-20:
I don't think that running multiple instances of cockpit on the same machine should be a problem. This would enable a situation where we can have a master computer running multiple instances of cockpit, each controlling its own microscope. It would also enable a testsuite where tests run in parallel, each starting its own independent instance of cockpit.

The issue is when multiple instances of cockpit connect to the same device. That should probably be prevented somehow, but maybe that's something on the device server?

iandobbie, 2022-05-17:
Good point David. However it seems to me that testing to see if we can bind to the server socket should eliminate this issue as each instance needs a different server socket. Looking at the code how this would work is not immediately clear to me. I will investigate