microscope-cockpit/cockpit #753: Multisite experiment not working

juliomateoslangerak, 2021-05-31:
Just merging from Micron/master. Multisiteexperiment is throwing while, I believe, trying to compose a file name template.
```python
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\mainWindow.py", line 122, in <lambda>
    ("Multi-site\nexperiment", lambda evt: multiSiteExperiment.showDialog(self) ),
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\dialogs\experiment\multiSiteExperiment.py", line 530, in showDialog
    dialog = MultiSiteExperimentDialog(parent)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\dialogs\experiment\multiSiteExperiment.py", line 230, in __init__
    configKey = 'multiSiteExperimentPanel')
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\dialogs\experiment\experimentConfigPanel.py", line 252, in __init__
    self.filepath_panel.UpdateFilename()
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\dialogs\experiment\experimentConfigPanel.py", line 567, in UpdateFilename
    basename = template.format(**all_mappings)
KeyError: 'cycle'
```

juliomateoslangerak, 2021-06-01:
So it seems to be that when the file name template is to be updated with date and time, in the case of a multisite experiment cycle and time are not provided and it fails. I suggest, inspired (or more copied) from pythons doc to use template.format_map in stead of format and provide a dict subclass that returns the key surrounded by curly brackets if the key is missing. 
This works for me fb11cdd6698349af481524cf52dccbd532892211

iandobbie, 2021-10-05:
I have pulled  this as it is needed to keep the multi-site experiment working. 
Previously it works once, then crashes on subsequent runs as the config file has the definition that breaks things.

This seems like a good fix

closed 2021-10-05 by iandobbie