microscope-cockpit/cockpit #765: dataSaver: wrong origin in DV files

dstoychev, 2021-08-17:
I noticed that all files saved from experiments are vertically flipped in ImageJ and OMERO. Digging through Bioformats revealed that they expect the origin to be lower left corner, whereas we write data "normally" with the origin in the top left corner:

https://github.com/ome/bioformats/blob/67960e5b5b1966a4879dddef258c6b89f96e69c5/components/formats-gpl/src/loci/formats/in/DeltavisionReader.java#L236-L246

Is there a specification of the DV format somewhere? I think we probably need to flip the data before we write it.

carandraug, 2021-08-17:
> Is there a specification of the DV format somewhere? I think we probably need to flip the data before we write it.

You could request the specification from GE but you had to sign an NDA (I never saw the specification because of that). But the DV files are for softworx which is a sort of "fork" of Priism. Priism uses mrc files, the base file format for dv. The Priism website has been down since UCSF remade their site but you can access it on web archive. The closest you'll have to its specification is at https://web.archive.org/web/20180729163334/http://www.msg.ucsf.edu/IVE/IVE4_HTML/priism_mrc_header.html which does not specify the origin. If I had to guess, I would say that the GE specification will also no specify it.

Are we sure the problem is how we write the dv file and not how we display the images in cockpit?

By the way, 10 years ago, bioformats (which omero and image use) would read the flipped upside down. I was the one that reported the issue, images were incorrectly displayed when compared to softworx. If you want to ensure that we do the right thing, I would check with softworx.

dstoychev, 2021-08-20:
>If you want to ensure that we do the right thing, I would check with softworx.
Excellent point. I made use of one of the DeltaVisions in the department. The summary is that the following three images have the same orientation -- eyepieces, Softworx, ImageJ. As such, I would say that ImageJ is correct and DV files do assume the origin is in the bottom left corner. We need to modify Cockpit to flip the data before writing it to the disk.
[dv_image.zip](https://github.com/MicronOxford/cockpit/files/7023404/dv_image.zip)

iandobbie, 2022-05-17:
This is relatively easily fixed if we decide to do that but we would need to heavily flag the change as all the SIM systems would have to respecify their stripe angles. Anther issue that would have to be dealt with internally to cockpit would be saving and load mosaics to make sure the correct transforms are performed on save and load. 

I cant think of any other immediate issues, mostly the orientation of the final image is pretty much irrelevant.