microscope-cockpit/cockpit #770: time points are split in different files when doing multiposition

juliomateoslangerak, 2021-08-20:
When I do timelapse experiments on multiple points, the different time points are saved in separate files. Either because the metadata is not expressing it or the naming convention is not compatible, bioformats is not getting that it is a single experiment. 
I have to manually import the images forcing bioformats to chain the time points and then it is working, but it is a very laborious process.
Would it be possible to append time points to previous files?

iandobbie, 2022-05-17:
This should be possible but as far as I know no one has ever made any effort to do this. I guess this is a "feature" of the fact that the multisite is effectively created by just looping through many single site experiments. It would need some logic and to ensure that it wrote the correct metadata at the start. Then decide to either keep the files open and loop through the file handles, probably easier but might run into resource problems if many sites are defined.

juliomateoslangerak, 2022-05-17:
While users are disturbed by the fact of having multiple files for a single position, In practice it turns to be more practical for post processing as some software packages do not load data lazily.
Maybe something to discuss when implementing NGFF.

iandobbie, 2022-07-14:
I think the additional complexity of preserving files across time courses with multi-position visiting is not worth the effort. I think once you do multi-point visiting many software solutions just dump each position and time point in a separate file.