microscope-cockpit/cockpit #774: Wrong sized opnegl panes are back with M1 Mac.

iandobbie, 2021-09-12:
Not sure exactly why but the M1 mac version of cockpit gives us the 1/4 sized opengl panels again. I will investigate. Maybe some of the code was switching on OS name which might have changed?

carandraug, 2021-09-13:
> [...] cockpit gives us the 1/4 sized opengl panels again. 

For reference, this issue is described on #628. The commit that fixed it is a5ce1c3522fab0f0a0a931bf8733785a47bfb1c7 (the commit refers issue #697 which is just the PR, the issue that fixes really is #628).

iandobbie, 2021-10-04:
Just another data point. IO confirmed that on an intel mac running macos 10.14 everything is fine but my m1 running 11.6 the opengl frames are too small. I assume that macos has moved to using different sets of variables in the canvas size mapping functions.

iandobbie, 2021-10-04:
Very strange outcome here. I wrote a quick script to start cockpit under pycharm 
import cockpit
cockpit.main(['cockpit','--no-config-files'])

This runs from the command line and produces the "correct sized" opengl frames. I think there must be some magic in the os which is treating different programs differently in the display drivers.

iandobbie, 2021-10-04:
Ok, there was some weirdness in my installed version of cockpit. Running 
python3 -m pip install --editable .
in the source directory installed it again and solved the weird opengl canvas size issues. I wonder if I had managed to get it running in intel eumlation mode or similar?

Closing as fixed as it appears that properly installing everything makes it work.

closed 2021-10-04 by iandobbie