microscope-cockpit/cockpit #775: Out of memory running multi-site timelapses

juliomateoslangerak, 2021-09-24:
There is an exception after some time acquiring multi-site experiments. Nothing crazy:
4 positions
180 cycles, one every 5 min
wide field z stack (1024 x 1024 x 21)

After cycle 74 files are 1 kb and there's where I imagine the problem started

The PC is windows, a bit old 2 sockets with a total of 8 cores and and only has 16 Gb of RAM but giving 5 min to handle the data should be enough. Anyway the memory is indeed filled up and to a good extent released, but only after a min or so, after aborting the experiment.

Here is the Exceptions trace

```python
Exception in thread processImages:
Traceback (most recent call last):
  File "C:\Users\julio\Applications\WPy64-3890\python-3.8.9.amd64\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Users\julio\Applications\WPy64-3890\python-3.8.9.amd64\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\imageViewer\viewCanvas.py", line 571, in processImages
    self.histogram.setData(newImage)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\imageViewer\viewCanvas.py", line 369, in setData
    these = np.bincount(np.digitize(data.flat, self.bins + i*h/m, right=True), minlength=nbins)
  File "<__array_function__ internals>", line 5, in digitize
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\numpy\lib\function_base.py", line 4860, in digitize
    return _nx.searchsorted(bins, x, side=side)
  File "<__array_function__ internals>", line 5, in searchsorted
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\numpy\core\fromnumeric.py", line 1348, in searchsorted
    return _wrapfunc(a, 'searchsorted', v, side=side, sorter=sorter)
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\numpy\core\fromnumeric.py", line 58, in _wrapfunc
    return bound(*args, **kwds)
numpy.core._exceptions.MemoryError: Unable to allocate 8.00 MiB for an array with shape (1048576,) and data type int64
Exception in thread processImages:
Traceback (most recent call last):
  File "C:\Users\julio\Applications\WPy64-3890\python-3.8.9.amd64\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Users\julio\Applications\WPy64-3890\python-3.8.9.amd64\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\imageViewer\viewCanvas.py", line 571, in processImages
    self.histogram.setData(newImage)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\imageViewer\viewCanvas.py", line 369, in setData
    these = np.bincount(np.digitize(data.flat, self.bins + i*h/m, right=True), minlength=nbins)
  File "<__array_function__ internals>", line 5, in digitize
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\numpy\lib\function_base.py", line 4860, in digitize
    return _nx.searchsorted(bins, x, side=side)
  File "<__array_function__ internals>", line 5, in searchsorted
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\numpy\core\fromnumeric.py", line 1348, in searchsorted
    return _wrapfunc(a, 'searchsorted', v, side=side, sorter=sorter)
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\numpy\core\fromnumeric.py", line 58, in _wrapfunc
    return bound(*args, **kwds)
numpy.core._exceptions.MemoryError: Unable to allocate 8.00 MiB for an array with shape (1048576,) and data type int64
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\safeControls.py", line 714, in OnTimer
    delayedresult.startWorker(self._onFetch, self._value.fetch, wargs=())
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\wx\lib\delayedresult.py", line 343, in startWorker
    thread.start()
  File "C:\Users\julio\Applications\WPy64-3890\python-3.8.9.amd64\lib\threading.py", line 852, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\safeControls.py", line 714, in OnTimer
    delayedresult.startWorker(self._onFetch, self._value.fetch, wargs=())
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\wx\lib\delayedresult.py", line 343, in startWorker
    thread.start()
  File "C:\Users\julio\Applications\WPy64-3890\python-3.8.9.amd64\lib\threading.py", line 852, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\safeControls.py", line 714, in OnTimer
    delayedresult.startWorker(self._onFetch, self._value.fetch, wargs=())
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\wx\lib\delayedresult.py", line 343, in startWorker
    thread.start()
  File "C:\Users\julio\Applications\WPy64-3890\python-3.8.9.amd64\lib\threading.py", line 852, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\safeControls.py", line 714, in OnTimer
    delayedresult.startWorker(self._onFetch, self._value.fetch, wargs=())
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\wx\lib\delayedresult.py", line 343, in startWorker
    thread.start()
  File "C:\Users\julio\Applications\WPy64-3890\python-3.8.9.amd64\lib\threading.py", line 852, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\safeControls.py", line 714, in OnTimer
    delayedresult.startWorker(self._onFetch, self._value.fetch, wargs=())
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\wx\lib\delayedresult.py", line 343, in startWorker
    thread.start()
  File "C:\Users\julio\Applications\WPy64-3890\python-3.8.9.amd64\lib\threading.py", line 852, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\safeControls.py", line 714, in OnTimer
    delayedresult.startWorker(self._onFetch, self._value.fetch, wargs=())
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\wx\lib\delayedresult.py", line 343, in startWorker
    thread.start()
  File "C:\Users\julio\Applications\WPy64-3890\python-3.8.9.amd64\lib\threading.py", line 852, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\safeControls.py", line 714, in OnTimer
    delayedresult.startWorker(self._onFetch, self._value.fetch, wargs=())
  File "C:\Users\julio\PycharmProjects\cockpit_venv38\lib\site-packages\wx\lib\delayedresult.py", line 343, in startWorker
    thread.start()
  File "C:\Users\julio\Applications\WPy64-3890\python-3.8.9.amd64\lib\threading.py", line 852, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread
```

juliomateoslangerak, 2021-10-10:
For that matter, a single site time-lapse experiment suffers from the same issue
[image]

juliomateoslangerak, 2021-10-11:
For info, the issue is reproduced in the production system, with real cameras and laser
[image]
s!

iandobbie, 2025-06-11:
I have finished my test and the system happily collected 100k frames and didn't appear to have any memory leaks. I think we need to commit some solution to this issue but I propose that we just delete the wait() and clear() calls because I think they block because of the CallAfter() call. 

I just tested without them and there didn't seem to be any specific issues. Data is being collected fine and you can still interact with the imageview, eg zoom, pan and change display ranges.

iandobbie, 2025-06-12:
No I haven't tried multi site experiments. My stage  controller has loud fans so I tend to just leave it off. I will turn it on a try with multi-site. 

As a temporary solution I suggest we pull the timeout solution and if testing is successful we just remove the wait and clear event code.