microscope-cockpit/cockpit #777: Multi position experiment crashes

juliomateoslangerak, 2021-10-04:
Trying to deal with #775 on a simulated environment I get a crash after a few time points. No memory issue like in #775.
Here are the steps to reproduce. Relevant config files attached
Environment:
Fedora 34. 
Microscope side:
Installed microscope master branch using pip and python 3.9
On the microscope  side I serve using deviceserver two SimulatedCamera's and two SimulatedLightSource
On the cockpit side I run on a conda installed cockpit (conda is the only way I could get wxpython on fedora). Again python 3.9 and the master branch of cockpit.
On startup everything seems to work fine. 
- Activate cameras but not the simulated lasers cause there is another issue with that.
- snap image to verify things work. They do.
- Mark 4 sites
- launch a multisite-experiment on the 4 sites witht default settings
  - 10 cycles
  - 60 sec per cycle
  - 0 waiting time
  - Z-stack experiment (4um, 0.15um z-spacing,...) no time for the lasers.
- And I click OK
The experiment starts fine but after timepoint 8 cockpit exits without error message. Remote micrsocope cameras and lasers seem to be completely fine end when I start cockpit again they respond well.
I attach the log files of cockpit as well as some config files.

The trace from the command line is this one:
```bash
(cockpit) [julio@localhost cockpit]$ cockpit --config-file Test_config.ini
Gtk-Message: 10:14:04.370: Failed to load module "canberra-gtk-module"
Gtk-Message: 10:14:04.370: Failed to load module "pk-gtk-module"
**
Gtk:ERROR:gtkcontainer.c:2743:gtk_container_propagate_expose: assertion failed: (child->parent == GTK_WIDGET (container))
Bail out! Gtk:ERROR:gtkcontainer.c:2743:gtk_container_propagate_expose: assertion failed: (child->parent == GTK_WIDGET (container))
Aborted (core dumped)
```
Interestingly, when I do the same experiment with bigger ROI and bigger Z-stacks, cockpit crashes earlier.
Also, running this in the windows configuration as in #775 the experiment continues much longer.

Con you reproduce? is there something else I can test?

[20211004_Mon-1014.log](https://github.com/MicronOxford/cockpit/files/7276703/20211004_Mon-1014.log)
[Test_config.txt](https://github.com/MicronOxford/cockpit/files/7276713/Test_config.txt)
[Test_depot.txt](https://github.com/MicronOxford/cockpit/files/7276715/Test_depot.txt)

[SimulatedCamera_127.0.0.1_8005.log](https://github.com/MicronOxford/cockpit/files/7276723/SimulatedCamera_127.0.0.1_8005.log)
[SimulatedCamera_127.0.0.1_8006.log](https://github.com/MicronOxford/cockpit/files/7276724/SimulatedCamera_127.0.0.1_8006.log)

juliomateoslangerak, 2021-10-04:
On Windows10, similar setup is going through the end of the experiment, at least up to the programmed 20 time-points
**Note: fix to #753 has to be applied.**
Can I assume it is an issue with Fedora and the installation through conda?

iandobbie, 2021-10-04:
I tried to setup a similar experiment on my macbook air m1 based, with 16GB of ram. It ran fine with 10 repeats of 9 positions (I got lazy and set a 3x3 grid of positions). Then I did 30 repeats which also worked fine, however my python porcess ended up taking 10.5 GB which is almost the size of the images on disk (11.9 GB) so I wonder if it just isnt closing the files and releasing the memory associated with them. I think unix (and by extension macos)  handles these types of issues much better than windows which hates processes with 10 GB + of allocated ram.

I think we are probably seeing the same behaviour just that macos doesnt crash yet. I think we need to investigate what the datalogger is doing in multisite experiment mode.

iandobbie, 2021-10-04:
I have played a bit with adding prints to the code as its not easy to run in the debugger without triggering timeouts. The code appears to be closing the files and triggering a garbage collection on the end of each time point at each position so I dont actually understand what the issue is.

juliomateoslangerak, 2021-10-04:
I think what you are describing is more related to #775 though I'm unsure it is not the same problem.
Is there an explicit call to collect the garbage?

iandobbie, 2021-10-05:
The GC call is in cockpit/experiment/experiment.py  somewhere around line 440:

        events.publish(events.EXPERIMENT_COMPLETE)
        events.publish(events.UPDATE_STATUS_LIGHT, 'device waiting', '')
        # Ensure the saveThread's memory, which includes all the images
        # collected thus far, is garbage collected. Otherwise memory tends
        # to pile up and then the GC has more work to do, which can interfere
        # with future experiments.
        print('before gc')
        gc.collect()
        print ('after gc')

I added the prints to see what was being run when. The debugger is almost useless in experiments due to the threading and timeouts for image triggering and data return.

juliomateoslangerak, 2021-10-10:
Same random crashes in Ubuntu 20 and 21